FreeCAD/FreeCAD #21439: BIM - windows not recalculated correctly after draft edit of the base  line of the hosting wall

zisco, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Windows are not recalculated after editing the line on which the hosting wall is based.
Steps to reproduce:
- create a Draft line
- create a wall based on the line
- place a windows on a face of the wall
- edit the draft line by moving one endpoint
- recalculate everything explicitly

The expected result be that the windows is placed on the wall.

https://github.com/user-attachments/assets/3fc823f0-9b81-4b79-b58f-f92654cc6fa0

### Full version info

```shell
OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.41726 (Git) AppImage
Build date: 2025/05/16 17:15:43
Build type: Release
Branch: main
Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/161.962/1
Installed mods: 
  * dodo 1.0.1
  * OpticsWorkbench 1.2.0
  * Reinforcement
  * Curves 0.6.62
  * FreeCAD_SketchArch
  * freecad.optics_design_workbench 0.6.2
  * sheetmetal 0.7.22
  * ExplodedAssembly
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-05-18:
Windows and doors added to walls do not take into account the change in wall geometry.
At the moment, there is only one way to stabilize this - expressions or the SketchArch addon (uses sketches). But it has its own problems. However, it will soon be more convenient to use.

https://github.com/paullee0/FreeCAD_SketchArch

https://github.com/user-attachments/assets/118edc12-1729-4b30-be8e-3e865c81aaa4

ErtugrulEge, 2025-05-18:
Windows and doors should use some   default expressions to follow wall position

paullee0, 2025-05-18:
@zisco  It seems you have installed the SketchArch add-on, if you use the Wall interactive tool to construct a wall and specify use Sketch as its base, it should create an ArchSketch.  Then, the feature @kaiwas indicates above should be available.

kaiwas, 2025-05-20:
> In which way is this issue a duplicate of [#14927](https://github.com/FreeCAD/FreeCAD/issues/14927)? I might be missing something, so apologies if it's obvious.

This problem is only indirectly related to the current one.
Yes - the standard methods for linking a window to a wall are only possible using expressions. But in turn, they are highly dependent on the numbering of vertices/edges/faces (if they were used in expressions)

The SketchArch addon partially solves these problems, but even there the connection can be lost if the original sketch line changes in the base sketch (for example, if you make a Split)

furgo16, 2025-05-20:
> This problem is only indirectly related to the current one.

So do I understand correctly that this issue should not be marked as a duplicate?

furgo16, 2025-05-20:
@maxwxyz given this feedback, can we unmark this issue as a duplicate?

yorikvanhavre, 2025-05-23:
This is a complex problem, because there are many ways a window could be "attached" to its host wall. In the current design, to avoid circular dependency problems (the window depends on the wall for its position, and the wall depends on the window for its opening hole), it's the window that keeps track of host objects. But it does not keep track of where the window is placed related to the wall.

We could add that, though, and therefore the window position could be changed when the wall changes.

The problem is, how to define parametrically where a window is positioned in relation to a parent object (which can be anything, not just a wall). And how to define that in a way that stays coherent with a change of a wall baseline. It's not a small problem.

Also, this is exactly the area of Assembly. There might be something there that could be useful, but it needs more investigation, and possibly more development of Assembly

kaiwas, 2025-05-23:
> which can be anything, not just a wall

Personal opinion - this is a categorically wrong approach. In most cases, the window is placed in the walls (conditionally 99.999% of cases). Another series of cases, when it is not tied to any object (autonomous).

It turns out that because of 0.0001% of cases, we discard what is really needed.
Personally, I do not know other architectural programs where windows can live separately from the walls. For example, in ArchiCAD or Revit, you cannot put a window in the air, it must be connected to a wall (and only a wall). But if you need to insert a window somewhere else, you can always use a library element (autonomous) and cut a hole. In any case, these are rare cases.

If this simplifies the task of placing windows - then my opinion is that we should limit the placement of windows only in walls. As is done in other programs, and this is good practice.

I do not have Revit at hand, but here is a demonstration from ArchiCAD.

https://github.com/user-attachments/assets/86a52052-6ece-49a5-a7c7-c06d828aed8b

kaiwas, 2025-05-23:
I would also like to add. Walls in other programs, even if they are "complex", are essentially separate segments. 
But the places where the walls are connected are processed in a special way. When two separate segments touch, their connection appears.

I don't know if this information will be useful, but it looks like this.
Maybe we should go this way?

https://github.com/user-attachments/assets/96eee118-a3ad-49f2-b60d-01c7a0281c65

kaiwas, 2025-05-23:
I have compiled a file with walls that are resistant to toponyming) But it requires doing a lot of unnecessary actions. (for example, attaching from PART. and changing window normals). But the main thing is that it works. Even after adding a second window, the first one does not drop out, because the number of vertices in one of the walls does not change).

Demonstration on video. You can literally see every step.

https://github.com/user-attachments/assets/4d127af1-f44a-47f6-b935-fd0ae0c71d11

paullee0, 2025-05-23:
Interesting ‘workaround’.  Try separate the edges in the sketch, add some new one - it may break?

The idea in ArchSketch is in some way similar to yours indeed - e.g. not using topologies in the Wall, it use the Geometries right in the ArchSketch, tracked and sort of TNP-tolerant thus.

kaiwas, 2025-05-23:
> Try separate the edges in the sketch, add some new one in between, reverse some direction of edges - it may break?

You don't need to check it - it will definitely break)

paullee0, 2025-06-13:
Except for ArchSketch-based ArchWall, there seems no plan to make other base objects work with this feature, right?

Should close this if so?  Thanks.

kaiwas, 2025-06-13:
> Should close this if so?

The problem is not solved "out of the box". Closing it would not be appropriate.

ErtugrulEge, 2025-06-13:
> Except for ArchSketch-based ArchWall, there seems no plan to make other base objects work with this feature, right?
> 
> Should close this if so? Thanks.

Dear, @paullee0, according to me these are the must have features.It's better to stay open, because, even if we close it, the same questions will come again.At least they can search and understand what the stiuation is.

paullee0, 2025-06-13:
I see, the question is more to the developers in fact, whether a Draft Wire particularly based ArchWall is possible to support the requested feature.

If not, the next question maybe whether Draft Wire should still be used as base in future if an alterternative TNP-tolerant worflow if available, and reduce confusion.

paullee0, 2025-06-13:
I have a look at the code.

If the Placement of the Host (Wall) is changed, the Children with Move with Host == True would be moved with same displacement.

However, editing vertex(es) of a DWire is not the same thing as changing Placement of the Wall, the placement of the Wall is simply not changed, so, nothing happens to the children / Windows.

paullee0, 2025-06-13:
Then, it is somehow what ArchSketch is basically doing (much more complicated under the hood of cos); I have ways to trace very edges in a (Arch)Sketch, but there is no hooks I found equally to trace edge in a DWire.

markkomartin, 2025-06-14:
> Except for ArchSketch-based ArchWall, there seems no plan to make other base objects work with this feature, right?
> 

I just discovered this method. I don't understand how it works but it works.

https://github.com/user-attachments/assets/97e6d3b9-6665-4684-9c6d-aad1c2d3d431

paullee0, 2025-06-14:
> I just discovered this method. I don't understand how it works but it works. If I move the wall afterwards everything breaks but that's another story.
> Transform.mp4

Below is about how it works, and how it won't.  The behaviour in the last paragraph is what this issue is about.

> If the Placement of the Host (Wall) is changed, the Children with Move with Host == True would be moved with same displacement.
>
> However, editing vertex(es) of a DWire is not the same thing as [...]

markkomartin, 2025-06-14:
> However, editing vertex(es) of a DWire is not the same thing as changing Placement of the Wall, the placement of the Wall is simply not changed, so, nothing happens to the children / Windows.

Thanks @paullee0

paullee0, 2025-06-14:
There are problems related to this issue - what are user want to change the length of the Wall, user need to stretch the Base DLine; what about user created a DWire say a rectangle and built a Wall, now he want to change the Length / Width of the Walls(Room), user need to edit the DWire.

Same problem.

kaiwas, 2025-06-14:
> Same problem.

Oh yeah.
I've seen many times that a good setup for one task becomes harmful for another task.
I'm really looking forward to the refactoring of walls and windows. But I'm worried that all this will work well with single walls, but poorly with walls of complex shapes (like squares). So for now my main hopes are tied to SketchArch. However, it has its own problems.

markkomartin, 2025-06-15:
>I'm really looking forward to the refactoring of walls and windows. But I'm worried that all this will work well with single walls, but poorly with walls of complex shapes (like squares).
> 
I also think that refactoring walls will work well only with single walls.
The " move base " problem will remain for walls with bases.

> Farewell parametrics! ((( That uncut diamond that only FreeCAD has.
> 
I'm not very good with sketch constraints. 
I feel like I'm wasting a lot of time setting up all the constraints and then I can only traslate a few walls.
I'm very scared of the topology problem.
That's why I'm not very attracted to parametric design at the moment.

paullee0, 2025-09-05:
> Windows and doors added to walls do not take into account the change in wall geometry. At the moment, there is only one way to stabilize this - expressions or the SketchArch addon (uses sketches). [...]

I am planning on 2 issues:-

1. Migrating  SketchArch TNP-Tolerant Parametric Attachment functionality to core FC Arch/BIM
2. Making Arch/BIM Objects independent of its Base object's placement 

Any comments?

ErtugrulEge, 2025-09-06:
>     1. Migrating  SketchArch TNP-Tolerant Parametric Attachment functionality to core FC Arch/BIM

This will be great  SketchArch has more functinality than core with migration for BIM development would be an important step

>     2. Making Arch/BIM Objects independent of its Base object's placement

I need an example to fully understand this statement but for  now I understand that there will be two objects: base line and wall object so when this relationship breaks you can move wall independtly but in 3D view there will be also its  base line object.If yıu have lots of wall ou will have lots of invisible  base lines and when you make them visible 3D view  will be very confusing.

[image]

Instead of this, I prefer when I move the object base lines can move with them in fact moving a wall means moving its base line 

[image]

furgo16, 2025-09-06:
> Making Arch/BIM Objects independent of its Base object's placement

As an alternative approach, I've been experimenting with a "Debase" macro I created, but it's pretty much in alpha state. I'll publish it in the forum soon. My idea was to have an easy way to decouple a wall from its base objects. @yorikvanhavre has a WIP that should implement the creation of **new** baseless walls, and my idea was to also be able to make **existing** walls baseless.

It works well with single-straight-line-based walls, but I'm not processing more complex walls than that. In light of this issue, need to see whether "debased" walls improve the situation with windows.

paullee0, 2025-09-07:
> I need an example to fully understand this statement but for now I understand that there will be two objects: base line and wall object so when this relationship breaks you can move wall independently but in 3D view there will be also its base line object.If you have lots of wall also will have lots of invisible base lines and when you make them visible, 3D view will be very confusing.

> Instead of this, I prefer, when I move the object, base lines can move with them, in fact, moving a wall means moving its base line

Indeed, there maybe  several common cases:-
(see screencaptures)

1. User has Sketch with wall layout (instead of individual wall segment) e.g.
- In general above Proposal (3) (4) apply
- e.g. a Base Sketch for 2nd floor is original at (0,0,0), Wall placement is (0,0,0)
- now the Base Sketch move to (0,0,3500), as Wall placement is attached to its Base with Attachment Support, Transform Link set to False, the Wall Placement is (0,0,3500) which reflect its 'actual' disposition (rather than currently 0,0,0)

2. (User may also just leave the Base Sketch at 0,0,0, and detach Attachment Support, just move the Wall Object but just which workflow is more intuitive to user, either works)
- (Wall placement is also 0,0,3500 reflecting its 'actual' disposition)

3. Ribbon Window in ScreenCapture 1, 2 below.  Base (Arch)Sketch of Ribbon Window could attach in placement to the base (Arch)Sketch / Wall layout
- Also similar to wall layout and above Proposal (3) (4) in general
- Window objects is attached to base Sketch (e.g. TransformLink=False, Attachment to base Sketch)
- So by just see these Sketch objects, users have a clear understanding of their relationship

4. Individual Door/Window + Links
- Now, base Sketch of individual Doors is more of a template of the Doors/Window + Links,
- its displacement has no bearing to the final model  (above Proposal (5) apply)
- Window's Link Transform set False, Attachment set to None
- Each Window / Links ' has their 'absolute' placement, moving their base Sketch has no impact

**SketchArch Add-on Features**
- With SketchArch, Individual Door/Window + their Links can already have individual displacement
- And parametric attributes direct each of them to place / move themselves
(e.g. 1st link of that Door move to 3rd Wall segment, 300mm from Start end, align to WallLeft)
(2nd link of that Door move to 5th Wall segment, 1000mm from Start end, align to WallLeft etc.)

 
[image]
[image]
[image]