FreeCAD/FreeCAD #21442: Spreadsheet parameters

bastl1, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

A spreadsheet is linked to the CAD-file (implemented). So user parameters should, can be selectable and referred to the shortest way so without " <<spreadsheet>>. " prefix but only by its pure name.
f.e.; instead of  "<<spreadsheet>>.my_value" only "my_value".
Also in the expression combobox, the user parameters should be presented ALL at first and other possibilities at last. 
This expression combobox should appear automatically when clicked on expression and the list in this combobox should shrink to the corresponding words when the user types in the first letter(s) (does not work accurate at the moment, f.e: L_value) .
All other parameters  that are rarely used or are FreeCAD internal parameters can be linked to with prefix, f.e.:  <<X-axis>> , <<Label>> . <<Sketch>>.

With an additional spreadsheet: Spreadsheet001.L_leg ( or spready-1.W_leg , [ name.name ] )

ComboBox->List:

1. places)  Parameters of the spreadsheet of this part and global parameters of a spreadsheet of the file
    global spreadsheet aliases can´t be used in local spreadsheets, they are auto extended by 00x on identical names.
3. places)  Parameters of the spreadsheet of other parts in this file
4. places)  Internal Parameters, Links, References, other possibilities 

This is priority - use friendliness.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-05-18:
> A spreadsheet is linked to the CAD-file (implemented). So user parameters should, can be selectable and referred to the shortest way so without " <>. " prefix but only by its pure name.
> f.e.; instead of "<>.my_value" only "my_value".

People use aliases like _width_, _height_, _radius_; automatic addition of them will pollute the global namespace.

> Also in the expression combobox, the user parameters should be presented ALL at first and other possibilities at last.

Even if their number is big? Some use too many variables at once.

> This expression combobox should appear automatically when clicked on expression

Why?

> the list in this combobox should shrink to the corresponding words when the user types in the first letter(s)

Its an expression completer, it works as it is supposed to work. The completer in _Chrome_ works in similar fashion.

Katt-Kushol, 2025-05-18:
Making it easier to find an alias from the spreadsheet could make the overall process more user friend. For example, if I start writing "BoxLength", can it be made possible to suggest all available parameters with that name in the suggestion, which will include the BoxLength alias from a spreadsheet although I have not typed in "Spreadsheet." in the beginning?

xtemp09, 2025-05-18:
You can rename Spreadsheet to something shorter.

I don't think this is the right way to deal with expressions. People use more than just one spreadsheet. When you type an expression, you reference an object and its properties. How the confusion with cell address can be solved? Each spreadsheet has cells A1, B1, C1 and other ones. Aliases also can have the same name, like _width1_, _radius2_.

----

Do you know about _VarSet_? [^1]

[^1]: https://wiki.freecad.org/Std_VarSet

maxwxyz, 2025-05-19:
@pieterhijma @FreeCAD/design-working-group @FreeCAD/cad-working-group what is the verdict here?

bastl1, 2025-05-19:
> People use aliases like _width_, _height_, _radius_; automatic addition of them will pollute the global namespace.
Exactly, software developers have to keep namespaces clean for the user 
> 
> > Also in the expression combobox, the user parameters should be presented ALL at first and other possibilities at last.
> 
> Even if their number is big? Some use too many variables at once.
That´s normal to parameterize drawings.
> 
> > This expression combobox should appear automatically when clicked on expression
> 
> Why?
Each user click is one to much and the user wants to open, use the combobox anyway. 
You even can implement the expression class into the dimension input window, because you know an expression anyhow.
And if a number is typed in the combobox disappears. And if a calculation is done it is stored as expression...
> 
> > the list in this combobox should shrink to the corresponding words when the user types in the first letter(s)
> 
> Its an expression completer, it works as it is supposed to work. The completer in _Chrome_ works in similar fashion.
Yes, it works, but the user wants mostly his parameters and pros know how to get internal namespace parameters.

xtemp09, 2025-05-19:
@bastl1, you seem like a typical user of one and only spreadsheet in the document. What if you have more than one spreadsheet? Would it be better to simply rename it with a short name?

bastl1, 2025-05-19:
> [@bastl1](https://github.com/bastl1), you seem like a typical user of one and only spreadsheet in the document. What if you have more than one spreadsheet? Would it be better to simply rename it with a short name?

Yes, I do, but parameters like " L_name "  for length_description are not sorted by typing in the " l " until the second sign is typed in " l_ " and with other overloaded letters it behaves strange, too.

xtemp09, 2025-05-19:
You just want to pollute the global namespace with local namespace of spreadsheet and want the completer to work with one letter instead of two.

All of these misfeatures are intended by design.

I would be glad if user-made parameters were in the top of the completer list, since user would like to use them first. All other stuff seems to be too far-fetched change of the design.

bastl1, 2025-05-19:
I see the point, and I agree. 
I´m a c++ programmer (gcc), QT, assembler (intel, AT&T), OOP.
I´m not a Python programmer.

Katt-Kushol, 2025-05-19:
If there are multiple spreadsheets in a file, and each spreadsheet has a variable with same alias, the suggestion list can show spreadsheet name in front of the suggested name, when a variable name is being typed in in the expression field. 
Consider the following example:

[image]

If I type in "Spreadsheet.Len" it suggests me "BoxLength" below the expression field. This method is working flawlessly, however, requires the user to type in the spreadsheet name first. 

Can it be programmed is such a way that when a user types in the variable name, the suggestion list will pull all the matching data from all spreadsheets, and show them in the following format:
SpreadsheetName.VariableName
The spreadsheet sourced variables can be placed on top of the suggestion list.

Will this add too much computation time to the current process? Is it justifiable for user friendliness?

FlachyJoe, 2025-05-19:
What is the argument for giving more importance to the Spreadsheet than to another object?
People who use VarSet for their main parameters (or user parameters on each objects) and spreadsheet only for BOM will dislike this.

Katt-Kushol, 2025-05-19:
> What is the argument for giving more importance to the Spreadsheet than to another object?

I did not think through when mentioned about who tops the suggestion list. I had the following image in my mind. 

[image]

Here, BoxLength, the intended variable appears at the bottom of the list. But, that is the only user-made variable, everything else on this list is kind of auto generated. I meant to say, in situations like this, the user-customized variables at the top makes more sense. It could be sourced in either VarSet or Spreadsheet.

FlachyJoe, 2025-05-20:
> I did not think through when mentioned about who tops the suggestion list.

In your issue description:

> ComboBox->List:
> 
>     1. places) Parameters of the spreadsheet of this part and global parameters of a spreadsheet of the file global spreadsheet aliases can´t be used in local spreadsheets, they are auto extended by 00x on identical names.
>     2. places) Parameters of the spreadsheet of other parts in this file
>     3. places) Internal Parameters, Links, References, other possibilities

Nevertheless I'm ok with

> the user-customized variables at the top makes more sense.

luzpaz, 2025-05-20:
Can someone propose a less generic more succinct name for this ticket that captures its essence ?

FlachyJoe, 2025-05-20:
@luzpaz "Reorder expression completer proposition to have user-added properties first"

bastl1, 2025-05-20:
luzpaz> Can someone propose a less generic more succinct name for this ticket that captures its essence ?

If you are sure it can be renamed without loses 
Should I rename it to:
"Optimization of Spreadsheet parameters" 
or
"User optimized Spreadsheet parameters"

bastl1, 2025-05-21:
I summarized all possibilities together with this image for better imagination and overview.

[image]

My wish would be to use parameters like this example:

{para1=100, ... }   // global,  lined to with the part lable as prefix
body1 : {para1=partlable.para1 - 50, para2=3, ... }
body2 : {para1=20, para2=4, ... },   expression. = body1.para1 - para1 = 30

pieterhijma, 2025-05-23:
> [@pieterhijma](https://github.com/pieterhijma) [@FreeCAD/design-working-group](https://github.com/orgs/FreeCAD/teams/design-working-group) [@FreeCAD/cad-working-group](https://github.com/orgs/FreeCAD/teams/cad-working-group) what is the verdict here?

I'm against giving preference to spreadsheets or varsets or both. There is also dynamic data for example that is an external workbench. In general, we don't know what properties users want to refer to. I do see value in improving the autocompletion to ensure that the suggestions are based on the property names, so for example "len" would suggest `<<Params>>.Length`, `Spreadsheet.Length`, `Box.Length`, etc. 

I propose to close this issue and create a new one "Gui: Suggest property names first in expression autocompleter".

pieterhijma, 2025-05-23:
> I summarized all possibilities together with this image for better imagination and overview.

Having the same labels for the spreadsheets in the image is already possible (with a preference checked), but I think it only leads to confusion. The tree view that the user typically sees is only one view of the objects. The objects are in general also in a "flat" namespace in the document. This is for example clear when you inspect the document objects that are in `App.ActiveDocument` in the Python console.