FreeCAD/FreeCAD #21446: There is no tool to facilitate the selection of interior entities

FEA-eng, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

One of the biggest problems with geometry selection system in FreeCAD is that there's no tool to easily select interior entities (e.g. faces or solids enclosed in other solids). The best solution would be a pop-up list like this:

[image]

Realthunder's fork already has it (it's called "Pick Geometry"):

[image]

But even a button to disable the default selection of the object closest to the screen and enable selection from subsequent proposals would make sense:

[image]

This issue is so common in FEM that a simple solution has recently been added to facilitate it (#18812). It's very specific since it only works for Compsolids and allows the user to choose between the different solids when applying e.g. materials but maybe it could serve as a base for development of a more general feature of this kind:

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41726 (Git)
Build date: 2025/05/16 17:15:43
Build type: Release
Branch: main
Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-05-18:
@FreeCAD/cad-working-group Please consider assigning a priority level to it. IMO this feature is essential for any CAD software.

pierreporte, 2025-05-18:
IMO it’s more UI/UX related. I classified it as P2 anyway.

maxwxyz, 2025-05-19:
Is it the same as https://github.com/FreeCAD/FreeCAD/issues/10308 ?
_Pick Geometry_ tool is implemented in RT link branch.

FEA-eng, 2025-05-20:
> Is it the same as [#10308](https://github.com/FreeCAD/FreeCAD/issues/10308) ? 

That one is about selection order priority and Pick Geometry is only mentioned in the comments while it deserves a separate issue due to its importance.

> _Pick Geometry_ tool is implemented in RT link branch.

Right, I mentioned it in the issue description. I hope it could be implemented in main (or something similar to it, maybe with less options).

kaiwas, 2025-05-20:
I know that everyone is tired of the constant mention of blender. But there are "centers" of faces in the grid mode. They are very convenient to select without any pop-up panels.
Just as an example.

This applies not only to planes. Depends on the mode (vertices/edges/faces)

[image]