FreeCAD/FreeCAD #21448: Thrashing when doing anything

srbaker, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When trying to add a rectangle to the attached file, FreeCAD thrashes and stalls.  GNOME pops up the "Force Quit" / "Wait" window several times.  Undoing does the same.

To reproduce: add a rectangle to Sketch001.  Then undo it.

This has been replicated on latest Ubuntu and openSUSE Tumbleweed, using FreeCAD 1.0.0 from Flathub.

Attached the system monitor screenshot as well.

[image]

(I cannot attach the FreeCAD file.)

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-19:
I am not surprised. A sketch with 1235 elements (328 of which are external geometry) and 1338 constraints will have this effect. I suggest you use different tools for this task. The Draft Workbench is probably a better fit. It has a working plane and a grid feature that can be used here. And I do not think you need constraints.

srbaker, 2025-05-19:
Aha.  Thanks for the explanation, it makes sense.  I'll close now.

closed 2025-05-19 by srbaker