FreeCAD/FreeCAD #21454: New bound object in new body shows strange colour behaviour

JDMils, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is about shapebinders in new bodies and how they appear when created.

Issue #1: New bound object in new body is the first item to be created.
Result: It and all subsequent shapes take on a yellow colour.

To recreate this:
* Create a new shetch in Part Design called Sketch_Original.
* Create a new body.
* Highlight Sketch_Original and create a new Shapebinder which appears in the new body as the first item created. The shapebinder has yellow edges.
* Pad the Shapebinder. The Pad is yellow transparent.
* All subsequent sketches created in the same body are also yellow transparent.

Is this normal?

If I highlight the body and toggle transparency, the edges remain yellow. If I repeat the toggle, the faces are now grey/solid however the edges are still yellow.

My topic thread:
https://forum.freecad.org/viewtopic.php?p=827630#p827630

### Full version info

```shell
https://forum.freecad.org/viewtopic.php?p=827630#p827630
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-23:
Please share an example file.
Is this also happening with SubShapebinder?

JDMils, 2025-05-25:
The issue seems to happen only with ShapeBinders not SubShapeBinders. I create a new body [BD01] and an object and pad it [OBJ01}, then add a new body [BOD02], select a face from OBJ01 and select the ShapeBinder, I get a copy of the face of the object in Body1.
This is where the interesting part starts. If I now create a new sketch in BOD02, I'm asked which attachment to use. If I choose to use ShapeBinderxxx, this is where the yellow edges and faces come in which is what i described above.
If however, I choose XY, XZ or YZ planes, everything works as normal.

If I keep creating sketches off the shapebinder and pad them, they all end up totally yellow, even if I select a non-shapebinder plane, such as the XY, YZ, XZ planes.

This ONLY happens if once the second body is created the very next step is creating the ShapeBinder from the previous body. If you create the second body and say create a sketch first, then import a shapebinder from Body01, the yellow issue does not happen.

[ShapeBinder inconsistency_JDMils.zip](https://github.com/user-attachments/files/20428903/ShapeBinder.inconsistency_JDMils.zip)