FreeCAD/FreeCAD #21460: BIM: IfcFurniture class is not loaded in FreeCAD

semhustej, 2025-05-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open IFC sample file in FreeCAD: 

[bonsai-furniture.ifc.zip](https://github.com/user-attachments/files/20300590/bonsai-furniture.ifc.zip)

- use these import settings

[image]

- the file contains 2 IfcFurniture pieces, they are not exposed in FreeCAD:

[image]

 

This is how the file looks like in IFC viewer or Bonasi:

[image]

The issue can also be reproduced by creating IfcFurniture class object in FreeCAD and than switching to Strict IFC mode. The furniture object disappears upon conversion to IFC.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41726 (Git)
Build date: 2025/05/16 17:15:43
Build type: Release
Branch: main
Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

I tested with few other IFC classes and the issue happens only with IfcFurniture from those, that I tested.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-06-19:
Skipping furniture is intentional. I assume because a big model with much furniture can be problematic.
https://github.com/FreeCAD/FreeCAD/blob/009056672473e5325d331c724253c3f25bc5db7f/src/Mod/BIM/nativeifc/ifc_tools.py#L936-L937

Removing that code results in furniture objects being imported, but without a shape.

semhustej, 2025-06-19:
My opinion is that at the current stage, there are quite a lot of bugs and unexpected behavior with IFC implementation in FreeCAD. I would rather see stable and expected, albeit slower behavior than more options for power users.

- I would enable `IfcFurniture` and `IfcFurnishingType` classes to load as any other class. Making such exceptions as not loading some classes confuses users. I have spent quite a lot of time figuring out what is happening :)
- If this is not acceptable, the option to enable loading of `IfcFurniture` and `IfcFurnishingType` classes should be present in the import dialogue, so that the user has a way to find out that these classes are not being loaded

[image]