FreeCAD/FreeCAD #21461: BIM: 1.0.1 walls cannot be copied

ErtugrulEge, 2025-05-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyone!
If you set "move base" property true, you can copy only the base line not the wall

Steps to reproduce:
1.Check Edit->Preference->BIM->Set "move base" property to True by default
2.Restart FC
3.Open a new BIM document
4.Draw a wall
5.Use BIM_copy

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

BIM

### Anything else?

This issue is only 1.0.1 specific, you can copy walls on dev version

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-20:
I can reproduce the behavior it FreeCAD 1.0.0 and 1.0.1.

As stated above, I am also  not able to reproduce the behavior in FreeCAD 1.1.0dev.41726.

luzpaz, 2025-05-20:
Are we backporting fixes or should we close this ticket ?

ErtugrulEge, 2025-05-21:
I don't know lots about development stuff but I think, if  it will take time to fix the 1.0.1 version , we should close the ticket.Instead of wasting time fixing things that have already been fixed, we should spend our time on more important things.This is not a critical issue.  I just reported it. I can wait the next version @maxwxyz

yorikvanhavre, 2025-05-23:
If it's fixed in dev version, then indeed the bug can be considered as fixed... I'll close this then! Reopen if needed!

closed 2025-05-23 by yorikvanhavre