FreeCAD/FreeCAD #21469: External geometry setect wrong feature

mgth, 2025-05-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In Sketch002, if I try to add an external geometry, it just select the wrong one.
[HC260-pulley.zip](https://github.com/user-attachments/files/20343293/HC260-pulley.zip)

I tried to open it with 1.0.1 and it does the same, but I did not try to fully make it in 1.0.1

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/79.7968/1.75
Installed mods: 
  * CurvesWB 0.6.12
  * Defeaturing 1.2.6
  * ExtremeProDark 2.6.4
  * fasteners 0.5.38
  * FreeCAD-Ribbon 1.9.1
  * FreeCAD-themes 2025.1.7
  * freecad-xr-workbench 0.9.6
  * freecad_streamdeck_addon 0.1.7 (Disabled)
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.4.2
  * workfeature
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-05-20:
Further investigation into the file reveals, the the Pad and the PolarPattern fails check geometry. Apparent cause of this failure is placement of of Sketch001. Once it is resolved, and the features OK'd by CheckGeometry, the external geometry performs as expected.
File attached.

[HC260-pulley_FR.zip](https://github.com/user-attachments/files/20348294/HC260-pulley_FR.zip)

luzpaz, 2025-05-25:
is this Part WB ?

Katt-Kushol, 2025-05-25:
> is this Part WB ?

Part wb is the default on my system, hench it showed part wb in the beginning of the gif. However, the sketch is within the active body when I tested, and later modified the sketch under pad. All the modifications were done in Sketcher.

FEA-eng, 2026-10-25:
I can't reproduce it now, but it would be good if someone else could check.

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```