FreeCAD/FreeCAD #21471: Buggy pad corrected by recompute

mgth, 2025-05-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The last pad is very buggy. The shape is recovered after selecting the whole tree, mark for recompute, and recompute.
[bug.zip](https://github.com/user-attachments/files/20343544/bug.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/79.7968/1.75
Installed mods: 
  * CurvesWB 0.6.12
  * Defeaturing 1.2.6
  * ExtremeProDark 2.6.4
  * fasteners 0.5.38
  * FreeCAD-Ribbon 1.9.1
  * FreeCAD-themes 2025.1.7
  * freecad-xr-workbench 0.9.6
  * freecad_streamdeck_addon 0.1.7 (Disabled)
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.4.2
  * workfeature
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-23:
Please describe the issue and steps to reproduce, could you add screenshots?
Did you try to reproduce this in safe mode or in 1.0?

Pendelf, 2025-05-23:
Indeed on the last extrusion there is an artifact, I don't know what it is connected with, but if in the first sketch switch Make internals to true it disappears. At the same time, if you switch it to false again later, the bug does not appear. Both sketches seem fine

https://github.com/user-attachments/assets/7261d4c1-4d2d-4780-b999-a719f29f68cb

brettowe, 2025-05-24:
I stumbled across this recently also i recreated the issue in a new file and that is attached as pre-broken
it should look like this:
[image]
if you toggle refine to true on last pad it will do this:
[image]

on the original file doing recompute did fix it, this one it seems to go into a new crazy design
so there might be another step required to make it happy again
I might of had multiple pads marked as refine: true 
instead of just doing 1 in this file

[broken-21471.zip](https://github.com/user-attachments/files/20427322/broken-21471.zip)

version info:
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/80.9812/1

```

FEA-eng, 2026-06-16:
In the second file, if I enable Refine and Allow Compound (it asks for it), I get this:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260609 (Git shallow)
Build date: 2026/06/09 14:33:40
Build type: Release
Branch: grafted,grafted
Hash: 9cb82b95e9f52f3ac2edc9d42525b39a1d77532a
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```