FreeCAD/FreeCAD #21472: thickness and fillet features depend on processing order

efferre79, 2025-05-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have created a simple body and I want to apply fillet and thickness features on opposite faces, the figure shows the current status after applying the fillet operation on the top surface:

[image]

and I want to apply the thickness feature on the bottom surface.

The result depends on the order of execution of the features. Both features use a 1 mm parameter (radius or thickness) . If I execute the thickness feature first there are no problems while if I apply the fillet first than it is impossible to get the expected results without tricks (e.g. reducing the thickness value to 0.99 mm). An exception is thrown 

```
12:16:54  <PartDesign> FeatureThickness.cpp(152): Exception on making thick solid: BRep_API: command not done
12:16:54  Thickness: Failed to make thick solid
```

### Full version info

```shell
OS: Gentoo Linux (XFCE/xfce/tty)
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Unknown
Branch: (HEAD detached at refs/git-r3/HEAD)
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.12.9, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.1, boost 1_85, Eigen3 3.4.0, PySide 6.8.2
shiboken 6.8.2, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.2606/1
Installed mods: 
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-23:
Please share the example file.
@pieterhijma I assume this is an OCC issue?

maxwxyz, 2025-06-11:
Could it be the same as https://github.com/FreeCAD/FreeCAD/issues/5829

pieterhijma, 2025-06-13:
@efferre79, The processing order is relevant for the operation. If you want the inside of the box to have 90 degrees edges, you first have to apply thickness, then fillet. If you want the inside of the box to have rounded "edges", then you first have to apply fillet, then thickness. In the latter case, the thickness will take into account the fillet for the operation.