FreeCAD/FreeCAD #21474: BIM: Unable to exit Strict IFC mode without error

semhustej, 2025-05-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- create `Arch_Wall` as a sample BIM object
- enter IFC strict mode
- click the `Strict IFC Mode` button again to switch to Hybrid mode
- error shows up

```
15:20:58  Traceback (most recent call last):
15:20:58    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41726-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_status.py", line 236, in on_toggle_lock
15:20:58      unlock_document()
15:20:58    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41726-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_status.py", line 348, in unlock_document
15:20:58      doc.removeProperty(prop)
15:20:58  RuntimeError: {'sclassname': 'class Base::RuntimeError', 'sErrMsg': 'property is locked', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'property is locked', 'btranslatable': False, 'breported': False}
```

- create new `Arch_Wall` object
- new wall is create as an IFC object instead of FreeCAD object

See behavior in the video:

https://github.com/user-attachments/assets/83895e44-9bcb-4104-bfab-7765eab57950

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41726 (Git)
Build date: 2025/05/16 17:15:43
Build type: Release
Branch: main
Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

I am not able to reproduce the issue in FC 1.0. The issue is quite new in the weekly builds (not older than 1 month).

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-05-20:
That's due to the introduction of the `locked` keyword for properties in https://github.com/FreeCAD/FreeCAD/pull/20778. The code tries to remove some IFC properties, but it's not possible, as they've been defined as `locked`, i.e. non-user-removable.

https://github.com/FreeCAD/FreeCAD/blob/d680de81c053d0342f8eace65b10ae880685069f/src/Mod/BIM/nativeifc/ifc_status.py#L345-L348

A solution is to remove the `locked` argument from the property definition for `IfcFilePath`, `Modified`, `Proxy`, `Schema`. All other properties under the IFC group as well.

Alternatively, @FlachyJoe is there a way to avoid users from removing properties in the UI, but still be able to remove them programmatically via the API?

pieterhijma, 2025-05-21:
> Alternatively, [@FlachyJoe](https://github.com/FlachyJoe) is there a way to avoid users from removing properties in the UI, but still be able to remove them programmatically via the API?

Yes, you can simply remove the locked status bit from a property, an example:

```python
>>> doc = App.newDocument("temp")
>>> # App.setActiveDocument("temp")
>>> # App.ActiveDocument=App.getDocument("temp")
>>> # Gui.ActiveDocument=Gui.getDocument("temp")
>>> doc.addObject("App::VarSet", "VarSet")
<App::VarSet object>
>>> doc.VarSet.addProperty("App::PropertyLength", "Length")
<App::VarSet object>
>>> doc.VarSet.getPropertyStatus("Length")
[21]
>>> doc.VarSet.setPropertyStatus("Length", "LockDynamic")
>>> doc.VarSet.getPropertyStatus("Length")
['LockDynamic', 21]
>>> doc.VarSet.removeProperty("Length")
False
>>> doc.VarSet.setPropertyStatus("Length", "-LockDynamic")
>>> doc.VarSet.getPropertyStatus("Length")
[21]
>>> doc.VarSet.removeProperty("Length")
True
```

Or to execute the script yourself:

```python
doc = App.newDocument("temp")
doc.addObject("App::VarSet", "VarSet")
doc.VarSet.addProperty("App::PropertyLength", "Length")
doc.VarSet.getPropertyStatus("Length")
doc.VarSet.setPropertyStatus("Length", "LockDynamic")
doc.VarSet.getPropertyStatus("Length")
doc.VarSet.removeProperty("Length")
doc.VarSet.setPropertyStatus("Length", "-LockDynamic")
doc.VarSet.getPropertyStatus("Length")
doc.VarSet.removeProperty("Length")
```

closed 2025-05-23 by yorikvanhavre

semhustej, 2025-05-31:
Tested with the latest weekly and it's fixed. Thank you.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25

```

PR #21488 (2025-05-23): BIM: allow deleting locked IFC properties when needed
When switching from native to non-native IFC mode in a BIM model, a number of IFC-related properties need to be deleted. Those properties are originally defined with `locked=True` so that the user cannot delete them via the UI. When they are attempted to be removed at runtime, the operation correctly fails with an error, since they are non-removable. Yet the expected behaviour is that removal works.

This PR effectively resets `locked=False` for those properties at the point of deletion, so that they can be removed programmatically.

Unrela [... cut]