FreeCAD/FreeCAD #21475: Fillet is not working well

paxal642, 2025-05-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello, 
I try to make a simple fillet but it makes a hole in the part.
Thanks for your help

[image]
[image]
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/Belgium (fr_BE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-05-20:
> ### Problem description
> Hello, I try to make a simple fillet but it makes a hole in the part. Thanks for your help

Can you attach the file you  are working on? Maybe a **simpler file that shows the problem** if you don't want to share your work to public.

luzpaz, 2025-05-20:
JFYI, Fillet bugs are known, most likely are related to OpenCasCade and have their own label: https://github.com/FreeCAD/FreeCAD/labels/Topic%3A%20Fillets

PeteHG99, 2025-05-20:
Thanks for providing the file. Looking at it, I'd have expected it to work. But it also fails for me on  Version: 1.1.0dev.41736 (Git) AppImage.

I think the fundamental issue with the model is, that you didn't apply a tangent constraint to the circle/straight line points.

It should still either work or give an error message and not result in a broken model.

paxal642, 2025-05-21:
Hi
Thank you @Katt-Kushol and @PeteHG99.
I have added the tangent constraint and it is working now !
@Katt-Kushol is there a list a best practice for freecad somewhere ?
have a nice day

maxwxyz, 2025-05-23:
I'll close this issue as it is working for oyu and there are already a lot of fillet related issues which are based on OCC

closed 2025-05-23 by maxwxyz