FreeCAD/FreeCAD #21476: BIM: prune groups from IFC export

furgo16, 2025-05-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The basic case of exporting a FreeCAD file that contains groups currently breaks object additions and substractions, rendering the exported file as unusable. See https://github.com/FreeCAD/FreeCAD/issues/18107

There is a [draft PR](https://github.com/FreeCAD/FreeCAD/pull/21441) that intends to fix that, but it does only fix the substraction part. The additions part is proving to be a bit more convoluted.

 If a user only wants groups in FreeCAD to logically group objects for easier browsing, but does not care about that specific logical grouping in the ifc file, would it not make sense to have an exporter preference to ignore groups upon export? That is, the exported ifc file would neither translate groups to IfcGroup nor IfcElementAssembly, it would simply ignore them.

I'm adding @semhustej's reply on the linked forum thread, as I feel makes much sense:

> In my opinion this would make sense. From the models created in FreeCAD BIM I have seen, the group usage usually does not resemble how IfcGroup or IfcElementAssembly are used in IFC.
>
> If there was an option to not use groups during IFC export/conversion, that would mean that in FreeCAD you could use groups freely, and at the IFC export/conversion they would not 'pollute' the IFC with unwanted IfcElementAssemblies.
If user actually wanted to have IfcElementAssemblies in the IFC project, he/she would have to adjust group usage in FreeCAD to this goal.
>
> I think this would be helpful for now, as it is first important to get the base case of IFC usage right.

FreeCAD forum discussion: https://forum.freecad.org/viewtopic.php?t=97111&start=10

OSarch forum discussion: https://community.osarch.org/discussion/comment/20488/#Comment_20488

### Full version info

```shell
1.1.0dev
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-21:
I would suggest for pruning of groups during IFC conversion to be set as default until a fix to [`FreeCAD groups messing with boolean operations during conversion to IFC`](https://github.com/FreeCAD/FreeCAD/issues/18107) is released.
At this point, most FreeCAD native IFC users are only starting to look at the functionality and a lot of them will encounter this issue and be confused by it.

furgo16, 2025-05-21:
Yes, I was actually thinking this too. Right now, adding groups is such a common (and often recommended) pattern in FreeCAD BIM models that I think mostly everyone exporting to IFC will end up with broken models. Which makes me wonder, given no more issue reports than the original issue, how many people actually use IFC in FreeCAD.

yorikvanhavre, 2025-05-23:
The main problem here is that groups are very useful in FreeCAD but not well translated in IFC (groups behave as layers and cannot be nested inside other objects). Note that we have an option to allow exporting nested groups, because I think it should be possible and wanted to try to convince IFC people ;)

So basically there is no 1:1 translation possible...

So yes adding an option to skip groups entirely is a good idea. Not sure if it should be the default though (my guess is that we'll get more people complaining of the absence of groups in exported file ;) )

furgo16, 2025-05-23:
So I think we all agree. I do find groups very useful as well, while creating the models in FreeCAD.

> So yes adding an option to skip groups entirely is a good idea. Not sure if it should be the default though (my guess is that we'll get more people complaining of the absence of groups in exported file ;) )

I'd stress what @semhustej already mentioned: the idea would be to make it default until https://github.com/FreeCAD/FreeCAD/issues/18107 is fixed. Right now, groups break the IFC export. In my case, I wrote a macro that removes all groups from the model, does the export, and readds the groups.

yorikvanhavre, 2025-05-26:
As always, a default is just a default. As long as it allows people to change it, I have no problem with changing the defaults :) Ok for me!

furgo16, 2025-06-08:
Thanks for the review! It looks like the parameter that stores the setting is either not being correctly saved or restored. I'll have a look.

PR #21583 (2025-06-03): BIM: ignore FreeCAD groups for IFC export, controlled by a user preference
## Summary

This PR adds a new feature to control whether a BIM model's standard groups are present or skipped when exporting that model to the IFC format.

The feature is controlled by a new IFC exporter user preference: `Export FreeCAD groups`. **Exporting groups is disabled by default**, meaning that groups (but not their children) will effectively be skipped from the resulting IFC exported file:

[image]

> [!NOTE]  
> - This PR changes the IFC exporter's current default behaviour, whereby now groups will NOT be pre [... cut]