FreeCAD/FreeCAD #21477: Assembly: Having the Assembly workbench disabled in the settings, and activating the Assembly by double clicking on the Assembly object causes a empty Workbench to be selected

luzpaz, 2025-05-20:
Having the Assembly workbench disabled in the settings, and activating the Assembly by double clicking on the Assembly object causes a empty Workbench to be selected.

[image]

2) Also, the Parallel joint stacks objects one over another, and this is annoying. Parallel plane does not mean they have to be centered. It also makes the assembly a more tedious.
 
https://github.com/user-attachments/assets/5a71eae7-78c7-4a7d-9345-48a2a67f9eb2
 
> ```
> OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
> Architecture: x86_64
> Version: 1.1.0dev.41726 (Git)
> Build date: 2025/05/16 17:15:43
> Build type: Release
> Branch: main
> Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
> Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
> shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
> Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
> Installed mods: 
>   * Assembly4 0.50.19-dev
>   * sheetmetal 0.7.22
>   * DynamicData 2.74.0
>   * Curves 0.6.61
>   * InventorLoader 1.5.1
>   * diff-tools 0.1.0
>   * ose-piping
>   * OpenTheme 2024.9.1
>   * fasteners 0.5.38
>   * osh-autodoc-workbench 0.2.3
>   * OpticsWorkbench 1.2.0
>   * lattice2 1.0.0
>   * kicadStepUpMod 11.6.9
> ``` 

 _Originally posted by @leoheck in [#15202](https://github.com/FreeCAD/FreeCAD/issues/15202#issuecomment-2894597109)_

maxwxyz, 2025-05-23:
@PaddleStroke FYI

PaddleStroke, 2025-07-16:
Well the same bug happens with double clicking a Body if PartDesign is deactivated no? And for a TD page if TechDraw is deactivated.

That bug does not happen with the tabbar as that widget is handling the case by adding a tab.

What would be the expected behavior then?

Imo pretty low priority since why would you deactivate assembly if you are using them.

leoheck, 2025-07-16:
It is a bug, it renders to an inconsistent behavior. Of double clicking an object loads its workbench, the workbench selector should not be empty. Even if the state of the workbench is to be hidden. 

Having an assembly in the file doesn't me the user is working with that thing. Maybe the file is from another person.

PaddleStroke, 2025-11-19:
Then this bug should not be marked for assembly but for core / combobox workbench selector. @luzpaz please update the ticket.

closed 2026-02-27 by kadet1090