FreeCAD/FreeCAD #21479: FEM: Running Elmer solver always freezes FreeCAD in the Windows weekly

FEA-eng, 2025-05-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the newest Windows weekly, running Elmer solver for any example (even a basic cube) freezes FreeCAD completely. This happens on at least two different Windows machines (with Windows 10 and 11). It was working normally in the previous weekly and there's no issue i 1.0.1 stable. Elmer log shown in the Solver Control panel shows that it's getting stuck at different phases of the solving process.

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marioalexis84, 2025-05-20:
Does it still happen if you run Elmer in single core mode?

FEA-eng, 2025-05-20:
> Does it still happen if you run Elmer in single core mode?

I currently have it configured only for a single core, I haven’t installed the multi-core version yet.

marioalexis84, 2025-05-20:
I experienced similar behavior a while back, but I didn't pay attention because I'm doing a complete refactoring of Elmer task panel.
Can you share more information about the issue?

FEA-eng, 2025-05-20:
> Can you share more information about the issue?

What details do you need? It happens with every single analysis whether it’s any FEM example or just a basic cube. It freezes at random stages and happens on two different computers only with the newest weekly so maybe it’s a fault of some seemingly unrelated commit. FreeCAD freezes completely so I have to terminate the process.

FEA-eng, 2025-06-23:
@maxwxyz Can you add a Blocker label to this one? Elmer became unusable on Windows so it's a nasty regression.

FEA-eng, 2025-08-07:
@marioalexis84 Could you have a look at this if you are available? This is the worst FEM issue now. I still can't run Elmer at all (any attempt freezes FreeCAD completely) even on Ubuntu. Of course, I will try to help with debugging but I've got no idea what could have caused it.

luzpaz, 2025-09-25:
@marioalexis84 do you think this will make it into 1.1 release ?

closed 2025-11-03 by chennes