FreeCAD/FreeCAD #21480: FEM: Instructions disappeared from Geometry reference selector in some task panels

FEA-eng, 2025-05-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Some FEM constraint task panels are missing texts regarding geometry selection. Here's what it looks like in the current Windows weekly for Body heat source:

[image]

And in 1.0.1 stable:

[image]

It's particularly bad in the case of Centrifugal load - now there's no indication which Geometry reference selector is used to pick a solid and which is used to pick an edge (axis of rotation):

- 1.1.0 dev:

[image]

- 1.0.1 stable:

[image]

It seems that all the long (multi-line) texts are gone and only the short ones are left:

[image]

Moreover, I have an impression that it happened when QT6 was introduced (I've noticed it with the Classic stylesheet that got severely broken then but now I realized it's also present with the Light theme).

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marioalexis84, 2025-05-20:
The geometry selector instructions have been moved to the tooltip to save space in the task panel. I don't think it should be long text in widgets.

>It seems that all the long (multi-line) texts are gone and only the short ones are left

No, those are C++-based task panels. Geometry reference selector is Python based.

marioalexis84, 2025-05-20:
We could add a line of text between the title and the button.

B0cho, 2025-06-04:
Hi all,
Is it still valid? What texts shall be added? Only which type of elements can be selected, or something more?

FEA-eng, 2025-06-04:
Yes, it's still valid but the question is how much information to add there. IMO at least that the first selector for Centrifugal load is for solids and the second one is for edges (to define the axis of rotation). But we could consider adding short selection tips to other constraints as well (like for the Fixed constraint) also for consistency.

B0cho, 2025-08-17:
> bumping for visibility and if this can be implemented in upcoming 1.1 ?
I will try to handle it before 1.1 freeze

marioalexis84, 2025-08-31:
I think a single line with information about the geometry type to select is sufficient.
I don't agree with filling the boxes with long explanatory text. That should go in the tooltip.
Regarding the difference between C++ and Python selection panels, we need to harmonize them at some point.

FEA-eng, 2025-09-01:
Yes, it should be sufficient. It's good to inform the user about the selection order (since it's inconsistent currently) but tooltips can be used for that. My main concern is with the FEM_ConstraintCentrif task panel since it has two geometry selectors and now only the tooltip indicates which one is used to select the solid and which one is used to select the axis of rotation. Bringing back the two sentences (one per geometry selector) like: "Select elements of type: Solid" and "Select elements of type: Edge" should suffice.
Just two remarks:
1. I think that it should say "element(s)" since it's not always necessary (or sometimes even not possible) to select multiple objects. Ideally, it should depend on the task panel and selector but it might be too much work for now.
2. Might be good to clarify that we are talking about geometry (not finite) elements here. Alternative short form: **Select geometry of type** (also resolves the first point automatically).

PR #23678 (2025-09-29): FEM: Restore and harmonize FEM Task selection hints
# The problem
Last contributions moved long details about geometry selection to tooltip, which is good solution.
However, no visible text at all makes dialogs aesthetic and not very convienient.
[image]

See #21480 for more details.

# Solution
Single line about type of selectable geometry was added. 
Longer details about selection are still shown in tooltip, when 'Add' button is hovered.

[image]

FEM Tasks, where geometry is to be selected, were harmonized to have same 'selection hint phrase'.

### Before
[image]

### Now
[image]

## Issues [... cut]