FreeCAD/FreeCAD #21485: This is a significant issue, which is particularly important for parametric design

gande007, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the techraw module, when there are multiple features in the view view, I annotate the size. When I hide one of the features, the view will rescale, causing the position of the features in the view to change, resulting in a change in the position of the dimension annotation. This is a major issue, which is particularly important for parametric design. I hope it can be improved in the next version. Thank you very much! #

### Full version info

```shell
freecad1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-21:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) 

- Test if the issue is still present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
- add full [About](https://wiki.freecad.org/About) info to the issue (`Help > About FreeCAD > Copy to clipboard`).
- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.
- provide screenshots

WandererFan, 2025-05-21:
Is the view Scale property set to autoscale?