FreeCAD/FreeCAD #21487: [Windows] Saving takes very long time on remote filesystem

knacky34, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm running FreeCAD on my work computer and saving files to a network filesystem (DFS).
It takes very long time to save files on this drive.

I ran a few tests with 1047kB project:
- Saving file to DFS -> ~1:15
- Saving file to local FS -> instantly
- Copying file from local FS to DFS -> 1s

With project containing a simple part (15kB):
- Saving file to DFS -> 3.5s
- Saving file to local FS -> instantly
- Copying file from local FS to DFS -> less than a second

With an empty project (2kB):
- Saving file to DFS -> 1.5s
- Saving file to local FS -> instantly

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * FrameForge 0.1.3
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Anything else?

I can't get many more details about the network/filesystem configurations.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-21:
@knacky34 would you be open to testing 1.1-dev and reporting back your findings ? 
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

knacky34, 2025-05-21:
@luzpaz, yes sure, I'll do it in following days.

luzpaz, 2026-01-11:
@knacky34 can you please retest on 1.2.0-dev ?

Also can you provide examples files for us to test as well ?