FreeCAD/FreeCAD #21489: Pocket partial not working

longtomin, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello,

I made a sketch with sockets and hole and nuts.
The sockets are a pad. The holes and nuts are pockets.
But one pocket (nut and hole are not calculated right.

I redesigned it in openscad, in the native openscad the rendering and everything else is fine.
I imported the openscad code in Freecad over the workbench, and even with this code the same "error" occures.

I tried with the weekly build and the official build.

![Image](https://github.com/user-attachments/assets/d7dc888e-da80-45c3-8

[PIBox_scad.txt](https://github.com/user-attachments/files/20366098/PIBox_scad.txt)

5de-6fe885c4e6e2)

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41736 (Git) AppImage
Build date: 2025/05/19 10:15:51
Build type: Release
Branch: main
Hash: 76ce8ccfd5b52e520c052d2193d7a244e83d4cb6
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * fasteners 0.5.35
  * OpenTheme 2024.9.1

----------------------------------------------------------------------------------
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * fasteners 0.5.35
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-12-19:
@KeithSloan can you reproduce ?

longtomin, 2025-12-19:
I downloaded the FreeCAD_1.0.2-conda-Linux-x86_64-py311.AppImage.
I used the FCStd attached to this issue.
It is still the same, so yes I can reproduce it as written in this issue.

luzpaz, 2025-12-19:
@longtomin can you test on 1.1RC1 ? https://github.com/FreeCAD/FreeCAD/releases/tag/1.1rc1

KeithSloan, 2026-03-20:
I note that visually the knut with the issue the hole in the middle does not go all the way through.

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

longtomin, 2026-10-25:
Hello,

not for me, it was just to show that there is a bug.

Regards