FreeCAD/FreeCAD #21493: Draft/BIM: Building US unit system - unable to insert unit type into taks panel input fields

semhustej, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When `Building US` unit system is selected, I am unable to insert any units (such as `'`, `"`, `mm`) into the task panel input fields.

Steps to reproduce:

- set unit system to **Building US**
- open a new file
- start `Draft_Line` 
- type in `10'` or `10mm` into the `Local X` input field
- only the number shows up, not the unit. This is an issue

[image]

This means that user is unable to insert feet-inches combination (like `5'10"`)

It is possible to use units when changing values in properties of objects.
I can not reproduce the issue with FreeCAD 1.0.1
The issue is relatively recent (circa last 14 days I think).

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-21:
There's another PR that I started testing last night #21147 which might have the fix in it already but I'm going to have to use two different distros in any testing as apparently the default `en_GB` Locale on Qt6.9 has some settings that I have never seen in the 8 years using Linux predominately Mint 19, 20, 21 and 22. I was testing the Draft Dimension switching between unit schemas and it was Building US that started errors for me but not for the author which is a little alarming.

Syres916, 2025-05-21:
@3x380V for me using #21147 this is still an issue, trying to create a line with specific keyboard entered sizes like 5' 10" should be possible, hopefully you can reproduce this issue or my testing is looking less and less useful.

3x380V, 2025-05-21:
@Syres916, yes, I noticed that using Unit converter. This is not yet fixed. Sometimes funny things happen, like trying to convert 42V to kV, which gives 42000000.00 mm^2*kg/(s^3*A) = 0.04 kV

3x380V, 2025-05-21:
@Syres916, to save you some time bisecting with respect to unit converter and inserting units:
-  ef6f60510ac13edb69e3c2f6bd23c03532a64124 works
- 1155f0d75281e9b9d70536d9b81dde7caececadd works as well

Minus, of course all that bugs that was fixed later. At least it is not the main culprit.