FreeCAD/FreeCAD #21495: Draft: Building US unit system uexpected behavior when inserting negative values

semhustej, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When inserting negative feet-inches values, FreeCAD calculates unexpected value.

Steps to reproduce:

- open new file
- start `Draft_Line` tool
- type `-10'5"` into the Local X input field
- FreeCAD recalculates the coordinate as `-9' 7"` . This is not expected.

As a workaround it is possible to type `-10' -5"` in this example. I am not from US, but I don't think that is usual way of inserting negative feet-inches units.
I tested BricsCAD and Bonsai BIM and it is possible to just type `-10'5"`

I am unable to test with current 1.1dev version, as there is currently different issue with Building US units.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Draft

### Anything else?

The issue was discovered by user in forum: https://forum.freecad.org/viewtopic.php?t=97182

I don't know if this is Core or Draft/BIM issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-12-07:
The issue is still valid.

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/2152cf0fa2c76524fd56d9d168892172c7603aa9

Negative feet and inches are read as one negative length, so -10 feet 5 inches becomes -125 inches. The existing -10 feet -5 inches input still works.