FreeCAD/FreeCAD #21496: Issue when enabling Stereo views, under Orthographic View

the-guyver, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I enable View > Stereo > Stereo Red/Cyan, the body displayed on screen would start flickering very fast when the body is viewed from certain angles, in Orthographic View.

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * PieMenu 1.2.7 (Disabled)
  * freecad.gears 1.3.0
  * woodworking 0.21.2.33771.0.24.1
  * A2plus 0.4.68
  * Manipulator 1.5.7
  * fasteners 0.5.38
  * lattice2 1.0.0
  * Telemetry 1.0.0
  * ExplodedAssembly
  * Curves 0.6.60
  * Assembly4 0.50.18
```

### Subproject(s) affected?

None

### Anything else?

I attached the FreeCAD project file that had the issue.

[stereo_issue.FCStd.zip](https://github.com/user-attachments/files/20371822/stereo_issue.FCStd.zip)

I uploaded a video to my Google Drive to demonstrate the issue, which can be viewed here:
https://drive.google.com/file/d/1fw1_h1P30gpLyNqH372c7blraM563r39/view

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ElementW, 2025-05-23:
I think I can confirm. Stereo makes orthographic stop being orthographic and the view matrices can get very weird at certain angles.

<details>
<summary>Version</summary>

```
OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.1.0dev.41726 (Git) AppImage
Build date: 2025/05/16 17:15:43
Build type: Release
Branch: main
Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.38
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.9.0.1
  * SearchBar 1.0.2
```

</details>

https://github.com/user-attachments/assets/d4cde0ff-fe69-4e7e-b23c-dc5ba4397577

JohnOCFII, 2025-05-23:
I was not able to reproduce the issue on either the 1.0.1 release, nor the a fresh development build on macOS on a 16" 2021 M1 MacBook Pro.

Curious what display rendering options the OP is using?

[image]

1.0.1 Release version:

[image]

Fresh development build:

[image]

```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Unknown
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * AddonManager 2025.5.1a
  * Telemetry 1.0.1
```

the-guyver, 2025-05-24:
> I was not able to reproduce the issue on either the 1.0.1 release, nor the a fresh development build on macOS on a 16" 2021 M1 MacBook Pro.
> 
> Curious what display rendering options the OP is using?
> 
> <img alt="Image" width="678" src="https://private-user-images.githubusercontent.com/7844212/447125782-d40d9f75-b05a-4a5b-a287-1ec05ac09481.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDgwNjQ0MjAsIm5iZiI6MTc0ODA2NDEyMCwicGF0aCI6Ii83ODQ0MjEyLzQ0NzEyNTc4Mi1kNDBkOWY3NS1iMDVhLTRhNWItYTI4Ny0xZWMwNWFjMDk0ODEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDUyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA1MjRUMDUyMjAwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTljMWMyNWE5ZjU4YmFlYWMwYzU5NmU4YWIxYjAwM2ViNWI3MmFjMTU0NmFlMGNlNzA2NTQ2OTg3ZWU3NmEzOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.7tstWHlnYTtf--Alw56JmCgD9SaUvIY3xfGxBTCdfXs">
> 1.0.1 Release version:
> 
> <img alt="Image" width="1545" src="https://private-user-images.githubusercontent.com/7844212/447126018-fe7cdb09-fcdd-4da3-8b0c-123f9e5422d7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDgwNjQ0MjAsIm5iZiI6MTc0ODA2NDEyMCwicGF0aCI6Ii83ODQ0MjEyLzQ0NzEyNjAxOC1mZTdjZGIwOS1mY2RkLTRkYTMtOGIwYy0xMjNmOWU1NDIyZDcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDUyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA1MjRUMDUyMjAwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmZlODAxMGMzMTU2NDljOTdlN2VkZTdlNWY0ODcxNTQ5NmMyNWY2NGI3OGU3ODQyYzlkNTFlNGUwYmY0OTdmNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.2N0JQm_P0Z8IXflMq9eSM35TOWPI8Rm7PWRQCvQTGTc">
> Fresh development build:
> 
> <img alt="Image" width="1545" src="https://private-user-images.githubusercontent.com/7844212/447126177-5fdfa6b8-78bc-413d-bda0-71946f0186c7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDgwNjQ0MjAsIm5iZiI6MTc0ODA2NDEyMCwicGF0aCI6Ii83ODQ0MjEyLzQ0NzEyNjE3Ny01ZmRmYTZiOC03OGJjLTQxM2QtYmRhMC03MTk0NmYwMTg2YzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDUyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA1MjRUMDUyMjAwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGIwN2RhN2ZmYTUwZGQ5M2M1ZmRlNDhkNzcwZTBlNWI1NDBlNTdlZDk4YjY0NWM5YjUzMzcxYTQ1ZWMzZjJiMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.mPYPCR-Lohusvlgh3QXXOpUcJ9IFuTq58MfmoVn-SWk">
> ```
> OS: macOS 15.5
> Architecture: arm64
> Version: 1.1.0dev.41837 (Git)
> Build date: 2025/05/23 13:10:01
> Build type: Unknown
> Branch: main
> Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
> Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
> shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
> Locale: English/United States (en_US)
> Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
> Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
> Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
> Installed mods: 
>   * OpenTheme 2024.9.1
>   * AddonManager 2025.5.1a
>   * Telemetry 1.0.1
> ```

Hi,
Here is a screenshot of my Display > 3D view settings

[image]

Also, I'm running the latest FreeCAD version, and I still have this issue:

[image]

Lastly, the machine I'm using is the Apple Mac Studio (2022)

the-guyver, 2025-05-25:
Note:
If the flickering does not occur for others after activating Stereo > Stereo Red/Cyan under Orthographic view, I suggest zooming in or out and orbiting around the object to see the "flickering" effect.

luzpaz, 2025-12-07:
@the-guyver can you still reproduce on 1.2.0-dev ? (weekly)