FreeCAD/FreeCAD #21498: <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1

bartoszp1992, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi!

I’m having weird trouble dimensioning a circle in the sketch. Here’s how you can reproduce the error:

https://www.youtube.com/watch?v=zT3FH9bJ3YA

### Full version info

```shell
20:50:10  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:10  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:10  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
20:50:10  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
20:50:10  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:10  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:17  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
20:50:17  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:17  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:19  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:19  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:19  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
20:50:19  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
20:50:19  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:19  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
20:50:20  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
20:50:20  <Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-21:
Full [About](https://wiki.freecad.org/About) info please.  Also zip up the .FCStd file and and drag it into github textbox to attach it to this ticket so we can reproduce it quickly.

Syres916, 2025-05-24:
It looks a lot like [#17266](https://github.com/FreeCAD/FreeCAD/issues/17266)