FreeCAD/FreeCAD #21502: Opening old projects with upgrated ver 1.0.1

pough, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Recently I've upgraded to 1.0.1 and after opening old project, which did not give any messages on previous version 1.0.0, on the new version 1.0.1 gives me messages (see attatched screen shot)

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.38
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-05-22:
> Recently I've upgraded to 1.0.1 and after opening old project, which did not give any messages on previous version 1.0.0, on the new version 1.0.1 gives me messages (see attatched screen shot)

Had you opened that file with any of the "newer than 1.0.0" weekly builds?  I **do** see that message when I open older files that I'd previously opened with a weekly build created since 1.0.0 was released.  I went back to a FreeCAD FCStd file that I hadn't opened since July, and I did not see that `App::Point` message.  Files created with something newer than 1.0.0 aren't guaranteed to open error-free with the 1.0.1 release.

luzpaz, 2025-05-22:
Did you back it up before opening in 1.1dev ?

closed 2025-05-22 by luzpaz

pough, 2025-05-23:
> Did you back it up before opening in 1.1dev ?
Yes, I've backed up the project, but also found, that opening the project with error messages, and then saving the project with 1.0.1 ver. helps. 
Next time it opens with no error.