FreeCAD/FreeCAD #21505: appimage downloader not foud

Vincenzoferrara, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

in the beta version there is no longer the app update button.

[image]

in the stable version it exists and works.

note: the Italian translation of "settings" does not work in the beta

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41736 (Git) AppImage
Build date: 2025/05/19 10:15:51
Build type: Release
Branch: main
Hash: 76ce8ccfd5b52e520c052d2193d7a244e83d4cb6
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.3233/1
Installed mods: 
  * AutoOpenSpreadsheet
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * btl 0.9.9
  * 3DfindIT 1.2.0
  * FeedsAndSpeeds 0.6.0
  * OpenTheme 2024.9.1
  * Assembly4.backup1718498214.0742533 0.50.12 (Disabled)
  * fasteners 0.5.38
  * TabBar
  * Assembly4 0.50.18
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-22:
Duplicate of #20590 the last I remember @looooo was looking to port the Updater to Qt6, unless I'm mistaken?

Vincenzoferrara, 2025-05-22:
And the latest version version. Lho downloaded yesterday