FreeCAD/FreeCAD #21514: ASME Inlined Dimension

desttinghim, 2025-05-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using ASME inlined dimensions in the Techdraw workbench, the area for the dimension text is much taller than it should be and the text is aligned to the bottom, see image below:
[image]
I first encountered this error running FreeCAD from the Arch package repository (v1.0.1). However, I also ran into the error after downloading the latest AppImage (1.0.1) and opening the drawing. I also tried remaking the part and recreating the drawing, but the problem persisted. I also downloaded the Flatpak. Only after opening the file in safe mode in the flatpak did the problem go away. My guess is that it is a regression in 1.0.1.

### Full version info

```shell
## Flatpak
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/

## AppImage
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-22:
It looks like the fix for this issue has been implemented in 1.1dev but not yet backported (ignore the A^ that's a Units issue that's still being worked on), screenshot using:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41810 (Git)
Build date: 2025/05/22 04:24:47
Build type: Release
Branch: main
Hash: 46a11b65389377799779825c5bedc644fcca80a9
Python 3.13.2, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.1, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9649/1
Installed mods: 
  * Silk 0.1.6
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

[image]

Syres916, 2025-05-22:
It would appear that this [commit](https://github.com/FreeCAD/FreeCAD/commit/999040803522e7b7ec3df6cc13c82baebe3f369d) is the fix , not sure whether it can be cherry picked or if a PR specifically for the _releases/1.0_ branch needs to be raised. @adrianinsaval could you try backporting the commit and let me know if I need to raise a PR please?

WandererFan, 2025-05-22:
I don't think we are doing backports to v1.0 anymore.  As I understand it, patches are going into v1.0.1 with a planned release date of "soon".

Syres916, 2025-05-22:
> I don't think we are doing backports to v1.0 anymore. As I understand it, patches are going into v1.0.1 with a planned release date of "soon".

It's the same branch for 1.0.2, 1.0.1 is already released.

luzpaz, 2025-05-22:
Are we closing this ticket ? (it would be good to have a label that indicates it's fixed in dev and if it will be backported or not)

Syres916, 2025-05-23:
@luzpaz I'm all for getting tickets closed promptly but the simple task of putting a Backport label on a PR failed in the first place (it's not in the slightest bit ambiguous that this was a fix and not a feature) so in this instance I'd like to see the backporting completed please before closure. It won't take me more than a few hours including testing to get the PR raised, it's just that cherry picking is a lot quicker.

I'm anticipating more than single figures of other PR fixes that weren't backported from December/January but finding them after the fact can only realistically be done by user feedback, the list of commits is just too large IMHO.

luzpaz, 2025-05-23:
Your point on the label does indeed seem correct. Just wondering how we can tweak the label system better so we can x-ref to locate our target query. I'm sure there is a way to do this.

Syres916, 2025-05-23:
The new PR template prompts the author regarding backporting but I don't see any significant change, even last week there was a PR that I'm quite clear should be backported but neither the author nor the maintainer wanted to add the label so what can I do????

maxwxyz, 2025-05-23:
My understanding is similar to @WandererFan that there are no more backports to 1.0, maybe this should be discussed at the next developer meeting.

closed 2025-05-23 by maxwxyz

Syres916, 2025-05-23:
Please ensure if the developers do decide that there should be no more backports to the `releases/FreeCAD-1-0` **branch** please create a topic in the _Announcements_ section of the forum so we all know, I read through the minutes of the last dev meeting and there was not even the slightest mention of it. Cherry picking a commit takes five minutes at most, if that's too long to wait then what can I say.

adrianinsaval, 2025-06-13:
backported e23a81e0c47f1240a562f69cf3220709f9ab0ff3