FreeCAD/FreeCAD #21520: DXF C++ Importer: Paper space objects imported even thought not 'Layout' not enabled in options

semhustej, 2025-05-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using DXF C++ importer, I get paper space obejcts (layouts) imported, even though the `Layout` setting is turned off in the import settings.

Steps to reproduce:

- open sample file (remove the .zip file extension)

[FC-DXF-layout-import.dxf.zip](https://github.com/user-attachments/files/20383967/FC-DXF-layout-import.dxf.zip)

- open new file in FreeCAD
- select `File` -> `Import`
- choose the sample dxf file
- in the settings, make sure the Layout option is not checked
- click `OK`
- the geometry of Layout (paper space objects) are imported. They should be excluded from the import

[image]

See behavior in the video:

https://github.com/user-attachments/assets/92c7e051-ad15-4f75-bc52-e43fbdb72ed4

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

File formats

### Anything else?

I am able to reproduce the issue in FC 1.0.1

As a workaround, Legacy Python DXF importer can be used.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-06-11:
For reference, we're talking about these objects, right? https://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-433D25BF-655D-4697-834E-C666EDFD956D

semhustej, 2025-06-11:
> For reference, we're talking about these objects, right? https://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-433D25BF-655D-4697-834E-C666EDFD956D

I guess that should be it.

From the FreeCAD point of view, Layout is something akin to TechDraw Page.