FreeCAD/FreeCAD #21527: AdditiveLOFT Fails for (physically) small construction details (and FreeCad  internally "Fails"). No problems observed for Big/Bigger construction details.

marek-stepan, 2025-05-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Iam working on "Air-Pipe-Duct"-component (to be 3D printed), and therefore Iam facing Critical problem of AdditiveLOFT feature Crashing and Failing => Blocking to finish my work. 
I would be very thanksfull to receive advice and feedback a) what Iam doing wrongly b) If there is workaround.

I have prepared DEMO FreeCad project to trigger Failure of AdditiveLOFT function Crash (because of smallconstruction Details). 
Small Construction details are unfortunatelly mandatory for my Design (it is detail how GLUE will be filled in gap between two components). 

Example FreeCAD project is TRIGGERING problem using Spreadsheet/Table.Variable (name = kill multiplier).
for Values/numbers Same as 0.98 there are NO problems, for Values/numbers smaller then 0.98 there is Crash.

For my Desing I want and I need usage of SMALL number (Target is 0.33 upto 0.6), and Examples for "Working" Multiplier 2 and Multiplier 5 are attached.  Those pictures shows only DETAIL of my Design, which triggers Issue. 

Observed symptoms:
* BRep_API : Command Not Done.  [See attachment]

To help with improvements (and analyzis with AdditiveLOFT function), I would like to additionally comment related Symptoms observed:
* NColecion_DataMap:find
* BRepFill requlation profile inconsistent
* Body Tipshape is Empty
*  Lost link during loading ( BUT Iam using healthy SSD disk!)
*  python Exception traceback 

For those additional Issues I do NOT have easy reproducible File, but I have just attached project where I have seen those troubles.  Please note, that I tryied recreating new sketches to avoid what I believed MAY trigger issue, but I did not created those problems intentionally.... (I still do not know WHAT exactly I did wrong.)

Thank you very much in advance for understanding. 

### Full version info

```shell
Problem confirmed on 2025/5/20 Released Weekly Build
Problem confirmed on 2025/5/13 Released Weekly Build

Official 1.0.1 Release Complains:
** Cannot create object 'Origin001': ('App::Point' is not a document object type)
**  <PropertyLinks> PropertyLinks.cpp(1013): Lost link to .........  while loading, maybe an object was not loaded correctly
** Extension is not a python addable version: 'Gui::ViewProviderGeoFeatureGroupExtension'
```

### Subproject(s) affected?

None

### Anything else?

[BUG-Report-DEMO-Please-use-killmult_Parametr_to_KILL_or_WORK_OK.zip](https://github.com/user-attachments/files/20400969/BUG-Report-DEMO-Please-use-killmult_Parametr_to_KILL_or_WORK_OK.zip)

[image]
[image]

[BUG-extra-addtional-problematic-project-example.zip](https://github.com/user-attachments/files/20401276/BUG-extra-addtional-problematic-project-example.zip)

[image]

[image]

[image]

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-23:
Please provide full about info.

@pieterhijma FYI

marek-stepan, 2025-05-23:
Issue/Bug was successfully observed on latest development version:
OS: Windows 10 build 18363
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
Installed mods: 
  * 3D_Printing_Tools
  * AddonManager 2025.5.1a
  * FreeCAD-Documentation-html 1.0.0-alpha
  * GDT
  * kicadStepUpMod 11.7.1
  * offline-documentation 1.0.0-alpha
  * toSketch 1.0.1
  * WB_Organizer 2024.1.29

And it was observed on same version WITHOUT any installed add-ons.  This version was NAMED in my text as Release from 2025/5/20 .
And I will update tomorrow PREVIOUS version identification (I will provide ABOUT.txt) which I have named 2025/5/13 (one week before, where I have started creating my constructions).

marek-stepan, 2025-05-25:
**Original attached file was created with FreeCad version (named by myself 2025/05/14 ):**

OS: Windows 10 build 19043
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.6656/1

**And same testing-tile shows SAME problem with latest available developer FreeCad verson (picture attached):**

OS: Windows 10 build 18363
Architecture: x86_64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25   

[image]

**Thank you very much in advance.**

marek-stepan, 2025-05-30:
Iam really sorry, CLOSing this issue was my mistake: Please I wish this Issue to be fixed. thank you in advance.

pieterhijma, 2026-10-28:
The provided model has problems at startup where the additive loft is not recomputed correctly.

I changed the value of `kill_multiplier` to various settings, 0.33, 0.6, 0.98. The model is not recomputed fully in any of those cases. I cannot detect any crash on FreeCAD main with OCCT 8.1 dev.

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 27.1.0dev.49321 +1 (Git)
Build date: 2026/10/27 12:27:31
Build type: Debug
Branch: adapt-to-occt-v8_1_0
Hash: 371a550ee620d900a88d2e96bf8cb9e47b7e45ec
Python 3.14.8, Qt 6.12.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.1, boost 1_92, Eigen3 5.0.1, PySide 6.12.0
shiboken 6.12.0, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 2.0.1, OCC 8.1.0.dev1-53706f78a8
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.5-arch1.1/AMD/AMD Radeon 780M Graphics (radeonsi, phoenix, ACO, DRM 3.64, 7.2.11-arch1-1)
```

One suggestion is to seek modeling help on https://forum.freecad.org/.

Since this issue is not a problem either FreeCAD developers or OCCT developers can act on, I'm closing it.

closed 2026-10-28 by pieterhijma