FreeCAD/FreeCAD #21532: [Sketcher] - External Geometry causes buttons on toolbar greyed out in FC 1.1.0dev (41214)

Katt-Kushol, 2025-05-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I noticed that the sketcher in FC 1.1dev version sometimes get the entire toolbar to greyed out, all the buttons become unavailable except the Dimension button. I suspect that external geometry may have something to do with it. I try to reproduce the situation in the video linked below at time markers: 1:46, 1:56, 2:09, 2:59, 3:18.
[https://www.youtube.com/watch?v=6lAnt1h_t6A](url)

Key Action Steps to Reproduce:
1. Have external geometry in a new sketch.
2. Try to pull on the external geometry randomly.
3. If it fails to show the issue, convert the external geometry to construction mode.
4. Repeat step 2.

I was also able to reproduce this issue in Safe Mode.

[image]
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Silk 0.1.7
```

### Subproject(s) affected?

Sketcher

### Anything else?

This issue was discussed in the forum, several experts weighed in, and recommended to report it as an issue.
Forum post: [https://forum.freecad.org/viewtopic.php?t=97143](url)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-23:
Does it work in the stable 1.0.1?
@PaddleStroke FYI

Katt-Kushol, 2025-05-23:
> Does it work in the stable 1.0.1?
Yes. 

[image]

I originally suspected it was being caused by an external geometry. However, on this sketch, there was no external geometry. So, we can rule that out as a possible cause for the bug.

Version Info:

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * Silk 0.1.7

Katt-Kushol, 2025-05-23:
Here the repeated instance on safe mode FC1.0.1 version.

[image]