FreeCAD/FreeCAD #21533: Dragknife dressup adds unnecessary maneuver

lagnat, 2025-05-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The original Gcode is:
```
G2 I2.344672 J0.246435 K0.000000 X2.232168 Y8.729204 Z0.000001
G2 I2.239044 J-3.509326 K0.000000 X4.488414 Y9.382617 Z0.000001
G1 X4.901217 Y9.392066 Z0.000001
```

The resulting dressup:

[image]

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41573 (Git)
Build date: 2025/05/05 17:45:01
Build type: Release
Branch: main
Hash: 3189030c11010631e726dbe350e9077dc84109bb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/macos
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-23:
@tarman3 any chance you can confirm this issue ?

tarman3, 2025-05-23:
I do not know nothing about and never use the tool Dragknife

lagnat, 2025-05-24:
To whoever maintains the CAM workbench, I'm reasonably sure this rewrite of getIncidentAngle will fix the issue.  That being said, I've only done little testing so far.

```
    def getIncidentAngle(self, queue):
        # '''returns in the incident angle in degrees between the current and previous moves'''

        angleatend = float(math.degrees(self.segmentAngleXY(queue[2], queue[1], True)))
        angleatstart = float(math.degrees(self.segmentAngleXY(queue[1], queue[0])))

        incident_angle = (angleatstart - angleatend + 360) % 360

        # The incident can never be greater than 180 degrees.  If it is
        # then we need to measure the other way around the circle.
        if incident_angle > 180:
            incident_angle = 360 - incident_angle

        return incident_angle
```

luzpaz, 2025-05-25:
Can you provide a PR, @lagnat ? Then folks can test your fix.

PR #21578 (2025-09-19): CAM: Dragknife dressup adds unnecessary maneuvers
The CAM dragknife dressup calculates the angles between subsequent paths and determines if it needs to maneuver the cutter around in order to make sharp corners.

The code calculates the vectors between any 2 movements (line vectors or arc tangent vectors) and subtracts them to determine how "sharp" the angle is.  Consider a vector at 350 degrees and the subsequent vector at 10 degrees.  The angle between them is 20 degrees however the existing code calculates it as 350 - 10 = 340 which is incorrect.  When the resulting calculation is greater t [... cut]
closed 2025-09-19 by sliptonic