FreeCAD/FreeCAD #21537: Wiki password reset does not send email

jclxx, 2025-05-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. The _wiki password reset_ function apparently fails to send email.
2. The _wiki initial password_ function apparently fails to send email.

* **How to reproduce**
  1. click link https://wiki.freecad.org/Special:PasswordReset insert username, leave email address blank, press "Reset Password", wait 5 mins (or 24 hours), check email, including spam, see none.  Result same if fill in email address, leave username blank.  Or both.
  2. (as a wiki admin) https://wiki.freecad.org/Special:CreateAccount create dummy account with temporary email address, check email
* **Unlikely to be only one user**: note that none of the new wiki users made in May have made any contributions.  This is consistent with initial temporary password email not being sent.
* **Might possibly be receiver blacklisting** My email account is at google.com, if FreeCAD outgoing email is blacklisted there it's still a FreeCAD issue.
* **Unlikely to be transient**: Tried several times over several days; checked spam folders; copy/pasted username to ensure accuracy; copy/pasted email address given to wiki admins.

### Full version info

```shell
This is ongoing wiki operations issue, not FreeCAD software.

https://wiki.freecad.org/Special:Version reports Mediawiki 1.40.1, pear/net_smtp 1.10.1
and many other libraries which could theoretically affect this.

Configuration information is unavailable to me.
```

### Subproject(s) affected?

Project Tools & Websites

### Anything else?

At a guess, the outgoing SMTP gateway is blacklisted somewhere.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-23:
Have you read?:
https://forum.freecad.org/viewtopic.php?t=6830

semhustej, 2025-05-23:
I think I can confirm the issue.

Yorik tried to create a new wiki editing account for me, probably on May 15th. I did not receive any email. I use gmail and did not find any email with activation options even in the spambox.

maxwxyz, 2025-05-23:
@kkremitzki FYI

jclxx, 2025-05-23:
Another new user confirmed to me they didn't receive the initial "temporary password" email.

Roy-043, 2025-05-30:
Related forum topic:
https://forum.freecad.org/viewtopic.php?t=97420

kkremitzki, 2025-07-12:
(For the record, the issue was due to a problem with our mail provider's IPv6 config.)

jclxx, 2025-07-13:
@kkremitzki **No change when I test it: apparently not fixed.**
@maxwxyz Can I suggest you reopen this?  With timezones, some of us had no chance to test after kkremitzki's work.

kkremitzki, 2025-07-13:
Hm, ok, I only tested the email send behavior itself, I'll look again.

jclxx, 2025-07-14:
I can confirm this worked for me just now: "forgot password" sent email to my gmail account in a second or so with temporary password, logged in, changed password, edited a page.

@kkremitzki Thanks for fix.
@maxwxyz can I suggest waiting for a few more confirmations before closing?  I see there's a backlog of about 50 created accounts which have never logged in, perhaps there are further issues.

maxwxyz, 2025-07-14:
Closing then. Please comment if there is still an issue

prokoudine, 2025-07-14:
Still doesn't work for me

closed 2026-01-21 by maxwxyz