FreeCAD/FreeCAD #21540: TechDrawings are messed up when opening files created with older versions

mpfusion, 2025-05-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There's a problem opening older files. The section views are incorrect and most of the dimensions are missing. Also FreeCAD outputs several errors in the console of the type “2D references are corrupt”.

This behaviour has existed since a long while. This means for the regular user that older files' TechDrawings need to be completely redone after a few FreeCAD update iterations. Most often the dimensions are wrong/gone. Sometimes the drawing itself is broken.

Creating a technical drawing for a complex part can take several hours. And from the user's viewpoint it's not acceptable to redo the whole drawing after a few FreeCADs update iterations for *every part* created previously.

Here I've attached an example created with `0.20.1`: [older_drawings.tar.gz](https://github.com/user-attachments/files/20408406/older_drawings.tar.gz)

**Version 0.20.1**
[image]

**Version 1.1.0**
[image]

The console shows:
````
DVS::getSectionCS - SectionView - failed to create section CS
<Exception> Dimension001 2d reference is a Line
<Exception> Dimension002 2d reference is a ArcOfCircle
<Exception> Dimension003 2d reference is a ArcOfCircle
DVS::getSectionCS - SectionView - failed to create section CS
<Exception> Dimension008 2d reference is a Line
DVD::getDimValue - Dimension009 - 2D references are corrupt (5)
DVD::getDimValue - Dimension009 - 2D references are corrupt (5)
DVD::getDimValue - Dimension010 - 2D references are corrupt (5)
DVD::getDimValue - Dimension010 - 2D references are corrupt (5)
````

# Current behaviour
Opening older drawings are messed up to the point that they need to be completely redone. Errors show up in the log.

# Expected behaviour
Opening older drawings should be possible and look reasonably similar compared to the older version. At least the views and dimensions should still match up.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.41815 (Git)
Build date: 2025/05/22 12:45:52
Build type: Release
Branch: main
Hash: aa5b8af36e13544b3e7333c7ec141e43d5be05e2
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.8
shiboken 5.15.8, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/Ireland (en_IE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0197/1
Installed mods: 
  * fasteners 0.5.38
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-23:
I'm afraid this has been the case since 2017, I have to keep multiple versions of FreeCAD including a 0.17 on Windows 7 in order to modify a customers model to get the Tech Drawing to update nicely. Recreating the models is just too risky and time consuming for me so that's how it's been. With 1.0.x there has been an agreement to keep the API locked down but that doesn't mean it won't be the same situation with 1.1.x. @WandererFan anything I've missed/not accurate?

mpfusion, 2025-05-23:
Keeping old FC versions around just to open previously created files is not practicable. What if the old version won't run on a current OS any longer? Will we have to maintain old VMs with OS and FC in sync?

I understand that older FreeCAD versions won't correct open files created with newer versions. But newer versions failing to open older files is just a nightmare from a user's point of view. This issue will make it hard to use FreeCAD in any sort of production enviroment.

benj5378, 2025-06-11:
It's very hard to ensure backwards compatibility. Though I do agree, that files should at least be upgradable and not be kaboomed at every version, making any archive with model totally deprecated.

I'm not sure what we can do much about the problem right now on the short run.

WandererFan, 2025-08-17:
The first thing I notice is that Body has geometry errors.  

[image]

The HLR algos are very sensitive to geometry errors.

WandererFan, 2025-08-17:
> I thought TNP has been worked on and was supposed to be mostly solved in FC 1.0.

You can't solve TNP, you can only reduce its impact.  In fact, there are 2 Topological Naming Problems: the one that has been worked on that addresses changes in Bodies, and a lesser problem which is connecting the output from the HLR and projection algos to the inputs.  The second one has not had the kind of concerted mitigation edit that the first has enjoyed.

WandererFan, 2025-08-17:
> What would be the right steps to dimension this correctly so it doesn't keep breaking?

If I need to dimension something while still making changes to the model, I would use references to the 3d model instead of dimensioning the 2d projected geometry.

Select the View, select 2 vertices in the 3d window, press the add dimension button and you should have a dimension that is somewhat hardened against TNP.

Side note: best to not use vertical or horizontal dimension button, as the program seems to get get confused over what is vertical/horizontal in 3d.

FEA-eng, 2026-10-25:
[image]

```
09:11:15  <Exception> Dimension001 2d reference is a Line
09:11:15  <Exception> Dimension003 2d reference is a ArcOfCircle
09:11:15  <Exception> Dimension002 2d reference is a ArcOfCircle
09:11:16  <Exception> Dimension001 2d reference is a Line
09:11:16  <Exception> Dimension002 2d reference is a ArcOfCircle
09:11:16  <Exception> Dimension003 2d reference is a ArcOfCircle
09:11:16  Base shape is not valid for boolean operationDimension001: Dimension001 2d reference is a Line
09:11:16  Dimension002: Dimension002 2d reference is a ArcOfCircle
09:11:16  Dimension003: Dimension003 2d reference is a ArcOfCircle
09:11:16  Base shape is not valid for boolean operation<Exception> Dimension002 2d reference is a ArcOfCircle
```

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```