FreeCAD/FreeCAD #21545: PART: The attachment does not remember local coordinates on first launch.

kaiwas, 2025-05-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I attach a window sketch to a wall plane. I assign references and specify local coordinates to shift the sketch.
But the OK and Apply buttons do not work.

To change the coordinates, I have to close the attachment and reopen it through the property.

Video demonstration.

https://github.com/user-attachments/assets/f8968962-63ea-4561-a446-989cc9652ed4

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41736 (Git) AppImage
Build date: 2025/05/19 10:15:51
Build type: Release
Branch: main
Hash: 76ce8ccfd5b52e520c052d2193d7a244e83d4cb6
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master.zip
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.62
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Pendelf, 2025-05-23:
I confirm, but with a sketch made in Sketcher works fine

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.62
  * fasteners 0.5.38
  * PieMenu 1.12.0
  * Road 2025.2.22
  * sheetmetal 0.7.22
  * Telemetry 1.0.1

maxwxyz, 2025-06-04:
is it the same in 1.0?

kaiwas, 2025-06-04:
Probably the command to launch "attachment" through the menu and through the properties of objects are different.
If I make an attachment through the properties, there will be no problem. However, for some reason when launching MAP MOD, the object refers to itself and I have to delete the reference. (maybe this is a separate problem)
But the problem is that not all objects have this property of MAP MOD. For example, BIM objects do not have them (until I press the attach button)

https://github.com/user-attachments/assets/d942c491-d300-4aad-963a-1612185e53a5

kaiwas, 2025-12-02:
> AFAICT this has been fixed.

I can't agree with you.

Note that I'm completely deleting the offset values ​​and entering numbers into the fields (for example, by double-clicking inside the field and selecting delete). This is important.
The millimeters appear visually, but it doesn't work.

https://github.com/user-attachments/assets/04a309ea-0fa3-4bc1-afef-47385a5721f4

But if I target the zeros and delete only those in the field (without deleting the mm), the offset will work. But that's very inconvenient.

https://github.com/user-attachments/assets/6d5be355-b79e-4002-980c-92d81bb736c0

Who ever thought of entering a measurement system where ONLY numbers should be?

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0rc1.20251125 (Git shallow) AppImage
Build date: 2025/11/25 17:53:08
Build type: Release
Branch: (HEAD detached at 5376760)
Hash: 53767601ea65de7cd9f6590e7d5856afc0841635
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1

```