FreeCAD/FreeCAD #21550: TechDraw renders App Links to PartDesign Bodies at wrong position if contained in two layers of Std Part containers with offsets

nagmat84, 2025-05-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

TechDraw renders App Links to PartDesign Bodies at wrong position if they are contained in two layers of Std Part containers with offsets. There is already a [FreeCAD forum discussion for this issue](https://forum.freecad.org/viewtopic.php?t=97132) and I have been able to create a [this MWE `TechDraw-Bug-With-App-Links.FCStd`](https://cloud.mhnnet.de/s/XBxdXLg9EjynM8R).

Consider the following assembly of struts:

[image]

The upper two layers of struts which form the small rectangles are contained in sub-assemblies. An StdPart Container contains one small rectangle of four struts and an AppLink duplicates the rectangle. Within the rectangle two edges are also App Links which references the actual PartDesign Body for the opposite edge.

See the following illustration

[image]

The local coordinate systems of the sub-assemblies are as follows

[image]

If I generate an TechDraw of the entire assembly ("FrameAssembly") as follows the result looks as expected

[image]

[image]

But if I select individual PartDesign Bodies which are a) within _different levels_ of StdPart container and b) the intermediate SdtPart container level is skipped, then TechDraw renders the inner PartDesign objects at the wrong position (or not at all). It seems that TechDraw somehow "forgets" to consider the offset of the skipped Std Part container.

For example, if one selects the lower two layers of struts for a TechDraw, the error occurs:

 - The PartDesign bodies of the lowest layer is within the outer Std Part container
 - The PartDesign bodies of the middle layer are within the inner Std Part container
 - But the inner Std Part container is not selected as the upper layer  of struts is not wanted 

[image]

[image]

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-24:
I guess that is expected as the placement is changed in the parent objects again? @WandererFan FYI

nagmat84, 2025-05-24:
What do you mean it is expected? In the sense that this behaviour is explainable from the code or in the sense that is is expected user experience?

For me the behaviour is surprising. In particular, if the AppLink object is replaced by a real PartDesign Body the TechDraw drawing is correct. In order for the buggy result to happen one needs two layers of nested AppLink objects. The rendering is as expected, ...

 - ... if the StdPart Container "Sub Frame Assembly 2" is a real SdtPart Container with AppLink objects only inside, _or_
 - ... if all four PartDesign Bodies "B-Profile x-y" inside "Sub Frame Assembly 1" are real PartDesign Bodies and only the StdPart Container "Sub Frame Assembly 2" is an AppLink

So there must actually three conditions come together:

 1. The project structure must contain (at least) two layers of AppLink objects such that there is an AppLink to a container that contains other AppLinks. (I call that "nested" AppLinks.)
 2. The linking container, i.e. the AppLink, needs an offset
 3. The TechDraw only draws some of the contained AppLinks not the entire container, i.e. the container is left out

For me this sounds like unexpected behaviour, because it conflicts with the idea that an AppLink object should behave like the linked object (re-usability) but with a different placement. It also prevents making larger assemblies with AppLinks to smaller assemblies which themselves use AppLinks inside. As a temporary workaround one must only use AppLinks on the lowest level (i.e. in leaf nodes) for bolts, nuts, screws, etc.

WandererFan, 2025-05-24:
Is this what we are trying to produce?

[image]

WandererFan, 2025-06-10:
In this case, TechDraw is faithfully reproducing the shape delivered by Part.  Fixing this is better done in App::Part or Part::Feature::getShape.

Using Python, this:
```
>>> # Gui.Selection.addSelection('TechDraw_Bug_With_App_Links_mwe','Part')
>>> # Gui.runCommand('Std_OrthographicCamera',1)
>>> ### Begin command Std_SendToPythonConsole
>>> ## Gui/CommandFeat.cpp(249)
>>> # try:
>>> #     del(doc,lnk,obj,shp,sub,subs)
>>> # except Exception:
>>> #     pass
>>> # 
>>> ## Gui/CommandFeat.cpp(251)
>>> # doc = App.getDocument("TechDraw_Bug_With_App_Links_mwe")
>>> ## Gui/CommandFeat.cpp(262)
>>> # obj = doc.getObject("Part")
>>> ### End command Std_SendToPythonConsole
>>> s = Part.getShape(obj)
>>> Part.show(s)
<Part::Feature>
```
Gives this:

[image]

[TechDraw-Bug-With-App-Links_mwe_Part.shape.FCStd.not.zip](https://github.com/user-attachments/files/20672922/TechDraw-Bug-With-App-Links_mwe_Part.shape.FCStd.not.zip)

WandererFan, 2025-06-11:
a) To me, there is a bug in Part::Feature::getShape.  It should not be returning a malformed shape.

b) The 3d window uses vertices from the object and it's children's mesh approximations to define lines and faces.  It is strictly a pictorial representation.  TD uses the object's actual shape and passes it through OpenCascade's Hidden Line Removal and projection algorithms.

nagmat84, 2025-06-12:
Understood. So a) it is an actual bug and b) the culprit is not TechDraw, but something in Part::Feature::getShape which is used by TechDraw but not by the 3D window such that the only victim is TechDraw.

Does this mean the badges "Needs feedback" and "Needs confirmation" can be removed?