FreeCAD/FreeCAD #21573: Core: Building US unit system does not work correctly with spinbox arrows

Roy-043, 2025-05-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The Building US unit system does not work correctly when using the arrows of quantity spinboxes.

1. Open the Preferences Editor.
2. Change the unit system to Building US.
3. Click OK to close the Preferences Editor.
4. Restart FreeCAD.
5. Open the Preferences Editor again.
6. Load the Sketcher WB if required.
7. Go to Sketcher > Grid.
8. The "Grid spacing" spinbox shows `3/8"`.
9. Use the up arrow of the spinbox.
10. Spinbox shows `99999999,00` (without unit).

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

3x380V, 2025-06-26:
@lupaz, this is fixed (probably by #21365). I cannot reproduce it and both up and down arrows are working (also tested Draft Rectangle case, no surprise here, the same editbox is used). Now it again behaves as in FreeCAD 0.21.2.

Roy-043, 2025-06-27:
Thanks. This has indeed been fixed.

closed 2025-06-27 by Roy-043