FreeCAD/FreeCAD #21575: BIM: Clone Level up broken (makeBuildingPart is not defined)

Harvie, 2025-05-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In BIM i click this:

[image]

And i do get this:

[image]

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

BIM

### Anything else?

Working in levels is not very well documented right now.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-25:
@Harvie can you provide a zipped file (drag it into github issue textbox to attach)? Also next time can you copy paste the error (instead of solely  screenshotting it) so if people search for the error it shows up in their search inquiry.
 
@semhustej any chance you can reproduce without a file ?

Harvie, 2025-05-26:
No need to provide file. Just use empty BIM project, create BIM level in it, set height of the level (eg. 300cm) and right click on the level, select copy level up.

```
02:10:21  Traceback (most recent call last):
02:10:21    File "/usr/lib/freecad/Mod/BIM/ArchBuildingPart.py", line 926, in cloneUp
    nobj = makeBuildingPart()
           ^^^^^^^^^^^^^^^^
02:10:21  NameError: name 'makeBuildingPart' is not defined. Did you mean: 'BuildingPart'?
02:10:21  Traceback (most recent call last):
```

closed 2025-05-26 by Harvie