FreeCAD/FreeCAD #21582: Quit FreeCAD may have change to cause the Window Manager Crash.

endysonx, 2025-05-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When quitting FreeCAD, it may have chances to crash the whole Window Manager. All window bars disappear including the task bar. This could happen in all versions of FreeCAD. I've tried from 0.20 to 1.1.0 dev latest master branch. They all have this problem with different chances. I've used FreeCAD for months, everything is good except this.  My OS version is opensuse 15.6,  my desktop environment is CDE, Common Desktop Environment. it's an old environment, but it works fine on other programs.

What is the detailed quit logic of FreeCAD ? Could anyone give me any clue on this. which part of code may likely caused this? I am trying to solving it. Thanks for any suggestions. 

### Full version info

```shell
OS: openSUSE Leap 15.6 ()
Architecture: x86_64
Version: 1.1.0dev.41791 (Git)
Build date: 2025/05/20 10:10:35
Build type: Unknown
Branch: main
Hash: e4238f58e13197dbcce66669f5079a317aee4dfd
Python 3.11.11, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, boost 1_88, Eigen3 3.4.0, PySide 5.15.12
shiboken 5.15.12, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.7.0
Locale: C/Default (C)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/94.662/1
Installed mods: 
  * Curves 0.6.62
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-31:
Could you try to restart in safe mode (Help>Restart in safe mode)?

endysonx, 2025-06-09:
I've tried the safe mode, it still crashes, However, I found that if I open the FreeCAD without opening any 3D design files and then I exit or restart in safe mode, it won't crash and just exit normally. It crashed only when I opened any 3D files and then exit or restart in safe mode (already in safe mode). the 3D file is a very simple 3D file without using any plugin, just sketch and pads from sketch.

Roy-043, 2025-12-05:
Another issue that is perhaps related: #25606.