FreeCAD/FreeCAD #21599: DLL load failed while importing ImportGui: The specified procedure could not be found.

karamellpelle, 2025-05-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I tried to import this step file: [BBC-microbit](https://resources.kitronik.co.uk/zip/step-BBC-microbit-V1.5.zip), but import fails:

```
23:34:30  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: DLL load failed while importing ImportGui: The specified procedure could not be found.
```

(No dialog is shown)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.9.0
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * IconThemes
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Anything else?

Importing works using a previous _Version: 1.1.0dev.40504 (Git) Conda_ dev version.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Pendelf, 2025-05-27:
It seems to import without any problems

https://github.com/user-attachments/assets/c7d6c899-1010-4c7a-9462-65a30d77cb90

karamellpelle, 2025-05-27:
Strange. Is your version the same weekly build as mine ([2025.05.19](https://github.com/FreeCAD/FreeCAD/releases/download/weekly-2025.05.19/FreeCAD_weekly-2025.05.19-Windows-x86_64-py311.7z))? I tried on another computer to see, but it gives the same error: `<class 'ImportError'>: DLL load failed while importing ImportGui: The specified procedure could not be found.`. This was the first run of FreeCAD on this computer (AFAIK).

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.9.0
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.192/1.25
```