FreeCAD/FreeCAD #21604: Create CAM Job triggers Unit warning

masterxq, 2025-05-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Create a new Cam job in the newest version will trigger the unit warning on my system.

All units should be configured correctly to cnc units (mm/min). Still the warning appears.

Documents created with an older Version seems not to have this problem. But if i open a file created with the latest weekly, older Versions will trigger the warning too.

https://github.com/user-attachments/assets/b6cc0f5f-6a3b-42c3-ac4e-2b20a09ac929

The file i saved in the video.
[unit_test.zip](https://github.com/user-attachments/files/20448327/unit_test.zip)

Possibly related to #19441

It could also be nice to know if i may ignore this.

### Full version info

```shell
OS: Debian GNU/Linux 13 (trixie) (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/05/13 09:39:01
Build type: Release
Branch: main
Hash: 7d57569eb493b929e5647bd9b0ad03b33d1acc45
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
```

### Subproject(s) affected?

CAM

### Anything else?

Possibly it could be worth it to check if this could be a language specific problem.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-05-27:
Tried file diagonale_messung.FCStd
not get any messages

luzpaz, 2025-05-27:
> Version: 1.1.0dev.14555 (Git shallow) AppImage

@masterxq that's not the _latest_ version. At the time of this writing revision 41866 is the latest release[^1]

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

tarman3, 2025-05-28:
Maybe this problem related to some local settings of OS

masterxq, 2025-05-28:
same problem. It needs to have something todo with the file.

[image]

luzpaz, 2025-09-26:
@masterxq can you help me understand where the issue is exactly ?

sliptonic, 2025-10-16:
The warning is intentional for non-conforming unit schemas.  The logic was improved in 
https://github.com/FreeCAD/FreeCAD/pull/24017

closed 2025-10-16 by sliptonic