FreeCAD/FreeCAD #21609: Core: Status bar hints: flickering can occur if another widget is collapsed

Roy-043, 2025-05-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There can be space issues on smaller screens for the status bar hints. When another status bar widget gets collapsed because of this, moving the mouse over the collapsed widget can cause flickering.

1. Switch to the Draft WB.
2. Create a Draft_Arc.
3. Select the edge of the arc in the 3D view. This will result in the display of info in the status bar that takes up considerable space.
4. Restart the Draft_Arc command.
5. Resize the FreeCAD window so that the Snap widget is collapsed ad only the grid button and an `>>` arrow remains visible.
6. Now move the mouse horizontally over the collapsed widget from left to right and back again.
7. Result the collapsed widget repeatedly gets expanded and then collapsed again.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-27:
@kadet1090 FYI

maxwxyz, 2025-06-04:
Please try in next weeks build if this happens again, might be fixed then.

maxwxyz, 2025-07-05:
I am not able to reproduce:

```
OS: macOS 13.7.6
Architecture: x86_64
Version: 1.1.0dev.42461 (Git)
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/72/1
Installed mods: 
  * freecad.gears 1.3.0
  * CfdOF 1.31.2
  * Defeaturing 1.2.6
  * CfdOF.backup1751209763.682869 1.30.5 (Disabled)
  * Gridfinity 0.11.9
  * Manipulator 1.6.0
  * fasteners 0.5.38
  * Telemetry 1.0.1
  * CurvedShapes 1.0.13
  * Cables 0.2.0
  * sheetmetal 0.7.23
  * FusedFilamentDesign 0.25.250
  * Curves 0.6.63

```

Roy-043, 2025-07-06:
Still present.
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42461 (Git)
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

luzpaz, 2025-07-21:
> status bar hints

@kadet1090 have you seen this issue ?

FEA-eng, 2026-10-25:
I can't reproduce it on Windows (can't even resize FreeCAD that much), it seems OS-specific.

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

Roy-043, 2026-10-25:
This has been fixed.

closed 2026-10-25 by Roy-043