FreeCAD/FreeCAD #21611: Feature request: Part Thickness with no faces selected

theosib, 2025-05-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is a feature request.

Let's say I have designed an axel in FreeCAD, and do that object I'll be attaching wheels. I also want to make a hole through some other body through which I will insert the axel. Due to printing tolerances, the hole needs to be larger than the axel, so I end up creating a whole 'nother object for the cut.

Often, in real designs, the positive and negative versions of the object are much more complex, requiring a great deal of work to make a version of an object that's just slightly larger for the cut.

Something that would solve this problem brilliantly would be to enhance the Part Thickness tool so that it doesn't require faces to be selected, resulting in an object that is not hollow but just a version of the original part but slightly padded away from all exterior surfaces.

This way, for the Union, I can select the original object in the hierarchy, and for the Cut, I can select the Thickness version that depends on it. This has the added benefit of automatically updating the thicker version whenever the original object is updated.

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/Classic/macos
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * lattice2 1.0.0
  * Alternate_OpenSCAD 1.0.0
  * Curves 0.6.50
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-27:
you should be able to do this with a subshapebinder which can be offset.

theosib, 2025-05-27:
Dumb question, but how can I use subshapebinder to offset all faces of a single object by the same distance along the normal to each individual surface?

theosib, 2025-05-27:
It turns out that the Part Workbench has a "3D Offset" tool that does what I want.