FreeCAD/FreeCAD #21615: BIM Tutorial some inline images are missing

jeremy-tools, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Just downloaded FreeCAD 1.0.1. New install, never tried this software before, of any version. Running the BIM Tutorial have noticed that many of the descriptive icons inline are missing, uniformly replaced with an icon of something like a 3D block object. Attached a screen sample showing the **Slab** icon appearing normally, while others do not. Note: so far, all the main (big) images appear fine. I have been following the browser-based tutorial so I can learn what tools these missing images are indicating, and they appear fine over there (https://wiki.freecad.org/BIM_ingame_tutorial using DuckDuckGo).

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

The Report View is empty when this problem occurs (no reports).

I searched for issues and found the existing (near duplicates) #17863 and #19090 but it isn't clear from either of those that this is the same issue.

Also, before I began the tutorial I'd made some adjustments to the application font (increased it by adding a python module to set fontsize a few points higher than default as I can barely read the application's fonts on my hi res monitor) and considered the possibility that may have been responsible, yet reverting back to defaults produced the same outcome. Also, reopening in Safe Mode produces the same outcome (missing icons).

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-28:
I am able to reproduce the issue in FreeCAD 1.0.1

I am not able to reproduce in 1.1dev version.

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

Roy-043, 2025-05-28:
I can confirm @semhustej's findings. Since the issue has already been fixed I'll go ahead and close this.

Fixed in:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

closed 2025-05-28 by Roy-043