FreeCAD/FreeCAD #21620: CAM: Array dressup does not set up tool controller correctly

knipknap, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using an Array dressup, the resulting path copies are generated correctly, but the controller state is not set correctly.

Example: If the job has a Profile operation (or any other, doesn't matter) and an Array, and the Profile has a controller set to use Mist, then the mist is disabled after completing the Profile operation and remains disabled for the rest of the array execution.

### Full version info

```shell
FreeCAD 1.0, but still present in current master.
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-28:
@knipknap please post full About info for each version you test on. It historically saves so much time for everyone. TIA

knipknap, 2025-05-28:
Yeah, this is probably not the dressup, but another array operation - sorry, I did not know there were two types of arrays.

```
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * btl 0.9.9
  * lattice2 1.0.0
```

Test file:
[test.zip](https://github.com/user-attachments/files/20481922/test.zip)

knipknap, 2025-05-28:
@tarman3 thanks for your example, which version is that for? I am asking because it looks corrupted when I open it in 1.0.1; the Profile operation is not within the job.

[image]

tarman3, 2025-05-28:
You can close this issue
This is old known problem with old Array

knipknap, 2025-05-28:
Ouch. Then I vote for fixing the issue with the old array, as I think the current behavior can even damage a machine.

tarman3, 2025-05-28:
Probably for this need to change something in all post-processors
Because it is the post-processor that inserts the commands of the change of tools and cooling

tarman3, 2025-09-09:
I trying to implement correct behavior of `Array` with several tool controllers and coolant
PR #23714

I get some positive result in this way, but also get some problems
I spend experiments with a code

FEA-eng, 2026-09-28:
There's a linked PR