FreeCAD/FreeCAD #21627: Adding an Assembly of Parts to another assembly causes all the binary tools used in the parts to reappear.

Maxwelldoug, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[Example.tar.gz](https://github.com/user-attachments/files/20493190/Example.tar.gz)

See ExampleAssembly.FCStd - creating a new file, creating an assembly, and importing the Example's assembly causes the cutting tool to suddenly resume existing.

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * OpenTheme 2025.5.20
  * woodworking 0.21.2.33771.1.0.1
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-31:
@Maxwelldoug could you please check if the issue is also happening in 1.1dev https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.05.26

Maxwelldoug, 2025-05-31:
> could you please check if the issue is also happening in 1.1dev
I cannot, unless 1.1-dev is available as a flatpak - my environment does not allow non sandboxed packages under any circumstance.

ryankembrey, 2025-06-05:
I have verified this issue in 1.1-dev

[image]

OS: Fedora Linux 41 (Sway) (sway/sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.42077 +1 (Git)
Build date: 2025/06/05 06:34:36
Build type: Release
Branch: add-spacing-preview
Hash: 234e147633fffe787282ae59495e229eeb02b262
Python 3.13.3, Qt 6.8.2, Coin 4.0.2, Vtk 9.2.6, boost 1_83, Eigen3 3.4.0, PySide 6.8.2.1
shiboken 6.8.2.1, SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.8.0
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * OpenTheme 2025.5.20
  * CfdOF 1.30.4
  * sheetmetal 0.7.22
  * fasteners 0.5.38
  * Render 2024.12.15
  * Telemetry 1.0.1