FreeCAD/FreeCAD #21628: BIM: The window does not make a hole in the cloned wall.

kaiwas, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The window does not make a hole in the cloned wall. However, it is visible in the tree view that the cloned wall is the HOST for the window.

Video demonstration

https://github.com/user-attachments/assets/05eb97f8-0e89-4f71-bb2f-85ec140c6829

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41866 (Git) AppImage
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OpenTheme 2025.5.20
  * FreeCAD_SketchArch-master
  * PitchedRoof
  * Test
  * sheetmetal 0.7.22
  * addFC-main 1.5.1
  * Road 2025.2.22
  * Reinforcement
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-05-31:
> Could you share an example file?

Okay. I'll do it later. But the whole process is shown in the video.
1) Create a wall.
2) Create a clone of the wall.
3) Insert a window.

kaiwas, 2025-05-31:
File. Remove the zip extension.

[CloneWall and window.FCStd.zip](https://github.com/user-attachments/files/20534550/CloneWall.and.window.FCStd.zip)

paullee0, 2025-06-03:
It seems a cloned wall is always identical to the original.

Try make a hole in the original - would the clone show the same opening.

If so, the question is whether a Clone should only respect the Base object of a Wall, and ignore all the addition and subtraction?

Or a few more switch to let users select alternative behaviour?

Things in BIM are getting more complex.

paullee0, 2025-06-04:
It seems an easy and available workflow is to simply 'encapsulate' the original Wall into another Arch Wall (or Arch Component), then the 'new' Wall has all the original features and behaviour of an Arch Wall object.

kaiwas, 2025-06-04:
> It seems an easy and available workflow is to simply 'encapsulate' the original Wall into another Arch Wall (or Arch Component), then the 'new' Wall has all the original features and behaviour of an Arch Wall object.

Maybe we need a new tool (specifically for walls)? )))
The cloned wall could contain all the windows and doors that were built into the original. Now they are just holes.

This would allow us to multiply identical floors (in a multi-story building) in a simpler way.

luzpaz, 2025-09-26:
Can someone confirm this issue ?

paullee0, 2025-09-26:
> > Can someone confirm this issue ?
> 
> The problem still exists. However, one change is that I can't wrap a wall clone with another wall (this workaround used to work). It's very easy to reproduce.

Is it an ArchWall  or Arch Component or what?  

Can share the file?

Thanks.

paullee0, 2025-09-26:
> It seems an easy and available workflow is to simply 'encapsulate' the original Wall into another Arch Wall (or Arch Component), then the 'new' Wall has all the original features and behaviour of an Arch Wall object.

Just tested, somehow it is not working this way, 'out of my expectation' :)

[image]

kaiwas, 2025-09-26:
> Can share the file?

All my actions are on video. I didn't save the file.
Create a wall, create a clone of the wall, insert a window into the clone.

(Without SketchArch)

paullee0, 2025-09-27:
> > It seems an easy and available workflow is to simply 'encapsulate' the original Wall into another Arch Wall (or Arch Component), then the 'new' Wall has all the original features and behaviour of an Arch Wall object.
> 
> EDIT - Re-tested, somehow it works this way - 'encapsulating the original Wall with another Wall', with some bug - needs to set width ? :)

OK, checked the codes.

Encapsulating a Wall with another Wall works in generally - except that the Window do not expect a Wall within another wall, so it just check the Width of the parent Wall the Width setting is probably 0, so it needs manually setting it to something the child Wall width say 300, so the Window punch a hole.

Would try a PR which check the recursively the descendant.