FreeCAD/FreeCAD #21629: Gui: Status bar hints misshapen (macOS / high-DPI issue?)

depthoffocus, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The status bar hints (cool feature!) are sort of distorted and double-wide on my Mac.

[image]

### Full version info

```shell
OS: macOS 15.5
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.75/2
Installed mods: 
  * ThreadProfile 1.96.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.22
  * Curves 0.6.62
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-28:
cc @kadet1090

depthoffocus, 2025-05-28:
> cc @kadet1090 
> 
> Edit:
> 
> > Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
> 
> Note: macOS here isn't showing what QtStyle is enabled. Usually we had it as 'unset' if it was. Maybe this is part of the problem ?

Is there a way to find out what that is so I can add it here?

luzpaz, 2025-05-28:
@johnocfii any chance you can weigh-in here ? Is this a QtStyle issue perhaps ?

depthoffocus, 2025-05-29:
> > [@JohnOCFII](https://github.com/JohnOCFII) any chance you can weigh-in here ? Is this a QtStyle issue perhaps ?
> 
> I built a fresh FreeCAD today. I'm not sure how to get the hints to show?

Go into Sketcher and use the polyline tool -- hints appear for that one at least.