FreeCAD/FreeCAD #21630: "BIM_unclone" is missing from the BIM panel or anywhere in the menu.

kaiwas, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

"BIM_Unclone" is missing from the BIM panel or anywhere in the menu. 
The only way to see this tool is to add it yourself to the user panel.

MY suggestion is to add unclone to a group that would open on click. See the picture.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41866 (Git) AppImage
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OpenTheme 2025.5.20
  * FreeCAD_SketchArch-master
  * PitchedRoof
  * Test
  * sheetmetal 0.7.22
  * addFC-main 1.5.1
  * Road 2025.2.22
  * Reinforcement
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-05-28:
@luzpaz  Sorry. I was inattentive. This command is available in the Utils menu. The open "Utils" menu is too big to be guaranteed to see all the necessary tools.

[image]

But still, I think that this team could be part of the "Clone-Unclone" group on the main panel.

luzpaz, 2025-05-28:
@kaiwas would you mind renaming the ticket ITC? 

cc @FreeCAD/design-working-group