FreeCAD/FreeCAD #21631: Canot place Windows or doors on Wall

pierre-bru, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a model that I want to modify. I need to add a wall which I did ;and a door .
I select the wall then select a door from the library set the proper parameters the shadow of the door appear but when i select the position everything disapear

[image]

. No door addition in the tree…

### Full version info

```shell
I added the screen capture above
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-29:
It occurs to me from the screenshot that h1, h2, h3 values are zero, which produces a warning when creating such a door or window.

The warning should be displayed in `Report view`:

[image]

If you are using `Opening Only` preset, you don't need to change these values from the default ones. The will be ignored, as opening uses only Width and Height.

semhustej, 2025-05-29:
I would recommend you to watch this video I created about placing doors and windows. It might help you avoid further pitfalls and understand how the placement works: https://youtu.be/kS8eftDsvbU?si=BvoMhqgOF58lUDLW

You should have attached the file with the first message, that way it's more probable you will get more helpful response.

luzpaz, 2025-05-29:
@pierre-bru in general, you need to attach file on github directly ([click this link](https://github.com/FreeCAD/FreeCAD/issues/21631)) because attaching via email will not work.

pierre-bru, 2025-05-29:
Thank you I Watched the tutorial very helpful
Pierre

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

pierre-bru, 2025-05-29:
Thank I will do it next time
Pierre

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

luzpaz, 2025-05-29:
I'm confused, is this just a workflow issue or is this a legit usecase bug that should be addressed, @semhustej  ?

closed 2025-05-29 by luzpaz