FreeCAD/FreeCAD #21633: BIM: Holes in roof are causing troubles

Harvie, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am modeling house that has semi-complex roof and it requires roof to have hole in it (walls protruding where the hole is, but being cut where the roof is). At first i was planning to just add subtractions to roof the way i would do with hole in a wall. But then i've noticed that such feature does not exists for roofs.

Later i've realized i can on the other hand convert any 3d object to roof, so i made pad with hole and converted it to roof. But i've noticed it's still not correctly subtracted from the wall:

[roof-hole.FCStd.zip](https://github.com/user-attachments/files/20496244/roof-hole.FCStd.zip)

[image]

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

BIM

### Anything else?

Also just having way to add "subtractions" to BIM roof would be easier.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Harvie, 2025-05-29:
@furgo16 guess i should've tried myself first, you are correct.

Also there is bit of a inprecision here, but it can probably be mitigated by changing the design by making the hole parallel with Z axis, therefore. im not sure if this is bug or not:

[image]

maxwxyz, 2025-05-31:
Closing this then as solved in 1.1dev

Harvie, 2025-05-31:
BTW when i add something to roof's subtractions, it correctly subtracts from the roof.
But then later when i add that roof to subtractions of some wall, it subtracts whole roof, ignoring the hole that was previously subtracted from that. Is that correct behavior? Seems like a bug to me...

furgo16, 2025-06-03:
> BTW when i add something to roof's subtractions, it correctly subtracts from the roof.
But then later when i add that roof to subtractions of some wall, it subtracts whole roof, ignoring the hole that was previously subtracted from that. Is that correct behavior? Seems like a bug to me...

I would suggest opening a separate issue for that one, and adding an example file.

paullee0, 2025-06-03:
> [@furgo16](https://github.com/furgo16) guess i should've tried myself first, you are correct.
> 
> Also there is bit of a inprecision here, but it can probably be mitigated by changing the design by making the hole parallel with Z axis, therefore. im not sure if this is bug or not:
> 
> [image]

I remember there is cornercase the alogrithm doesnot identify the surface to form the subtraction - the inner elliptical surface (some point's normal facing downward, some poinnt's normal facing upward probably confuse the algorithm).  

Can have a closer look later.

paullee0, 2025-06-13:
Anyone has idea on below or recommend some others ?  @maxwxyz ?

1. Ways to detect if a surface/shell has every point of it pointing either upward or downward? or
2. When a surface is extruded, if the solid itself is self-intersecting?

Thanks 

> > [image]
> 
> I remember there is cornercase the algorithm doesn't identify the surface to form the subtraction - the inner elliptical surface (some point's normal facing downward, some point's normal facing upward probably confuse the algorithm).
>

Roy-043, 2025-06-13:
Depending on the desired outcome is may work to subtract 3 volumes:

1. Extrusion of bottom faces in +Z.
2. The roof itself.
3. Extrusion of the top faces in +Z.

There may be co-planar issues though.

paullee0, 2025-06-15:
Somehow, the odd shape in previous discussion works if the extrusion is not tested - [Sketch based Arch_Roof and wall substraction](https://forum.freecad.org/viewtopic.php?t=84389&start=30)

Anyway, would use Roy_043 approach in the meantime.

('Odd' curve surface in the middle extruded with odd shape)
[image]
[image]

It doesn't work when the extrusion is teste

[image]