FreeCAD/FreeCAD #21652: Hole Cut Type Diameter unchangeable unless re-edited

ShaiV, 2025-05-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating a threaded hole, and selecting a Hole Cut Type (for example Countersink) the Diameter parameter is limited by a value derived from Depth parameter. However, when you close the Hole Task Window and reopen it, you can set the Depth parameter to whatever value you desire without the aforementioned limit. (I personally would like to have the Diameter parameter unlimited but not sure if that is the right approach)

### Full version info

```shell
1.00
1.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

alfrix, 2025-05-30:
this feature has been refactored in the dev version, could you check the behaviour on the [weekly builds](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)?
because i think this is not an issue anymore

ShaiV, 2025-05-31:
It appears you are correct.
Thanks!

closed 2025-05-31 by ShaiV